Mu2e/Offline #56: Tags did not propagate

kutschke, 2019-11-12:
Earlier today I updated the fork of Offline in my github account to the head of master. I tagged the head of master as v7_5_5; I merged in Ray's e19 changes and then tagged again as v7_5_6.  Then I requested a pull of my fork to Mu2e/Offline/master.  

I just checked out Mu2e/Offline and the tags are missing.

Does anyone know what I did wrong and how to fix it?  Maybe it does not make sense to expect that tags on a fork will survive a merge? 

I can fix it by cloning directly from Mu2e/Offline, tagging and pushing.  I think that I have permission to do that.

rlcee, 2019-11-12:
I believe this is designed behavior.  I can't find an explanation
why on the github site.  I did find this
https://help.github.com/en/github/administering-a-repository/creating-releases
Which I think is the intended way to tag.

Ray Culbertson  mu2e/FNAL  rlc@fnal.gov  630-840-6744  WH9E (MS122)

ryuwd, 2019-11-13:
Hi Rob,

I believe that Pull Requests are only designed to transfer commits from a fork, so the tag would not be pulled. I think this is deliberate to avoid tags from users’ forks ‘clashing’ with the tags on the main repo.

If you have Mu2e/Offline as a remote on the clone of your fork, you might be able to use 
`git push upstream --follow-tags` as a quick fix (since you have write access to upstream), but only if the tag is annotated. Otherwise, Ray’s solution is best.

Related discussion: [https://stackoverflow.com/questions/12278660/adding-tags-to-a-pull-request](https://stackoverflow.com/questions/12278660/adding-tags-to-a-pull-request) 

Best wishes, Ryun

kutschke, 2019-11-13:
Thanks Ryun,

  I followed Ray's solution.  After my initial surprise I figured out that avoiding tag clash was
a smart choice for the default behaviour.  Ray's solution worked  easily.

   Welcome to Mu2e.  We you at the last collaboration meeting?  I don't think we have met yet.
 

     Rob

ryuwd, 2019-11-13:
Thanks Rob! Unfortunately I haven't been to any collaboration meetings yet. I am an MSc student at Manchester, and my project is in cosmic tracking and tracker alignment in collaboration with David, Sophie, and my supervisors Mark and Marco. Happy to be on board!

Best wishes, Ryun

kutschke, 2021-04-15:
Resolved long ago.  Closing now.

closed 2021-04-15 by kutschke