Mu2e/Offline #60: Cosmic Track Selector

ryuwd, 2019-11-20:
- [ ] Select high momentum based on observable parameters
- [ ] Select tracks most useful for alignment and calibration
- [ ] Tracker coordinate accessors

sophiemiddleton, 2019-11-23:
I'm going to start looknig at the first option here - is this any overlap to you?

ryuwd, 2019-11-24:
Hi, there’s no overlap. I’m not actually working on this.
I opened this as an example for my demonstration for the CompSoft talk. Go ahead!

Is this issue related to the cutting functions you mentioned in the email thread on Friday?

sophiemiddleton, 2019-11-24:
Ok great, yes, I think I have resolved some issues that some of the hits flagged as outliers are not removed such that then the DOCAs dont always refelect the "good" hits. I've fixed that now and the ambig is back at around 97%. I have moved the diag list as well. Going to do some work tomorrow on trying to get the code in nicer shape and then I'll start looking at this. I want some advice from Dave as to how to make this more consistant with the Kalman, I have some ideas but need to check with him as he is the expert who wrote all that Btrk stuff.

Excellent, will let you know if anything else springs to mind :)

ryuwd, 2019-11-24:
I’ll wait for you to push those changes before doing any more work using the drift fit.
Since the MLE drift fit is appears too slow for calculating residuals for alignment, I’m going to l try to get familiar next week with BTrk and the Kalman fit code so I can try and help with that

ryuwd, 2019-11-24:
Also, is the “ high momentum based on observable parameters” selection something that would be based on track fit chi2 or similar? Since multiple scattering effects would worsen residuals and so the chi2, or something along those lines? Marco suggested plotting the Multiple scattering angle estimate (using truth momentum) vs. reco Chi2 to see the relation

However it seems like you already make a cut anyway on chi2 and I’m not sure how to calculate the MS angle since I need to access the radiation length of traversed volume (unsure how to do this in Offline)

sophiemiddleton, 2019-11-24:
Yes, I mean this is were the anlaysis side of things comes into it so we need to find features of the tracks which correlate to high momentum to an extent that we can make some intelligent assumptions that tracks with feature X will be higher momentum and/or will therefore produce worse fits.....MCS is of course one result of this so an interesting study as Macro points out

closed 2020-09-29 by ryuwd