Mu2e/Offline #86: Resolve TrackerConditions<->ProditionsServe dependency loop

rlcee, 2019-12-27:
Both TrackerConditions and ProditionsService need to use fcl config header files, such as:
TrackerConditions/inc/StrawResponseConfig.hh
TrackerConditions rightly depends on ProditionsService, but the reverse
should not occur, and this config file is the only way it does.
The object is acting like a utility container.  This could be resolved two ways:
1) move config headers to ProditionsService
2) move them to a lower library, either 2a) Mu2eInterfaces (already used
for the Proditions handle header) or 2b) a new directory, ProditionsConfig, for example.

Rob and Dave, let me know what you think

brownd1978, 2019-12-29:
Putting TrackerConditions/*Config.hh in Proditions centralizes the configs,
but those really belong in a tracker-specific package.  When other systems
copied that pattern ProditionsService would become an overpopulated mess.
Same problem moving them to a lower library.  We could create TrackerConfig
and move them there, but the simplest solution is to just live with this as
It's only a compile-time dependence and not a true loop.  That also keeps
the directory count down, and keeps the files connected to the code that
uses them.

kutschke, 2020-01-02:
Hi Dave,

   To my mind there is a good reason to get rid of compile time header loops - it is part of the
program to support pgit and to enable smaller partial checkouts.  I would like us to be using
this routinely by the summer.    For major development efforts pgit is probably of marginal use
but we need to do a better job of enabling smaller development efforts or it will be more
difficult to build the workforce.

        Rob

brownd1978, 2020-01-02:
Hi Rob,
pgit is an important use case.  I don't think people doing proditions
development will want or be able to use it though, so this particular
dependency won't limit us.  Of course it is simpler to have a blanket
policy (no compile loops).

I don't like having all the conditions headers for the experiment in a
single directory, that will be a mess and hard to maintain.  To break the
dependency we then need to bifurcate the proditions packages; one for the
end-user classes (like StrawResponse), the other for the proditions
headers.  If you and Ray are happy with that design then I will accept it
too.
Dave

kutschke, 2020-01-02:
> 
> Hi Rob,
> pgit is an important use case. I don't think people doing proditions
> development will want or be able to use it though, so this particular
> dependency won't limit us. Of course it is simpler to have a blanket
> policy (no compile loops).

That's what I would like.  We can enforce that programatically as a part
of validation.

> 
> I don't like having all the conditions headers for the experiment in a
> single directory, that will be a mess and hard to maintain. To break the
> dependency we then need to bifurcate the proditions packages; one for the
> end-user classes (like StrawResponse), the other for the proditions
> headers. 

I fully agree.

rlcee, 2020-01-03:
fixed in PR #92

closed 2020-01-03 by rlcee