Mu2e/Offline #89: CalPatRec helix chisquared has kludge factors

brownd1978, 2019-12-29:
The Chisquared for the helix found by CalHelixFinder includes a kludge factor which biases the comparison to the chisquared of helices found by TrkPatRec.  The kludge factor should be removed and the algorithm parameters adjusted accordingly to provide more accurate selection of the best helix.

rlcee, 2022-08-22:
Do I recall correctly that this was addressed recently?

brownd1978, 2022-08-22:
Namitha's change to MergeHelices bypasses CalPatRec chisquared calculation
in that context.  I don't know if it's used in other contexts.

NamithaChitrazee, 2022-08-24:
Yes, we had solved this issue at the MergeHelices stage (PR #812). The bias in the helix selection is removed as the chi squared calculation is done uniformly for all helices irrespective of TrkPatRec or CalPatRec in the MergeHelices module. But, I have not made any changes to the chi squared calculation method (LsqSums method) followed by CalPatRec.

rlcee, 2022-08-24:
Thanks. Would it appropriate to close this issue, or do you want to leave it open to remember to make changes to the CalPatRec chi2?

brownd1978, 2022-08-24:
I think its OK to close this issue

NamithaChitrazee, 2022-08-24:
Yes, I think it's OK too.

rlcee, 2022-08-24:
Thanks everyone!

closed 2022-08-24 by rlcee