Mu2e/Offline #97: Define best practices for using objects from Proditions.

kutschke, 2020-01-09:
Define best practices on the wiki; scrub the existing code and what for issues during code reviews.

Here some initial thoughts on best practices.

Proditions returns shared_ptr<T>.  We should not pass this down the call chain to where it is used.  At a minimum we should derefence it to a T const& once at the start of produce and pass that down the call chain.  Even better, if T has a lot of structure, extract a const& to the lowest level object that is needed to do the job - then pass that down the call chain.  It's OK to plan for future evolution and pass a const& to a higher level object than is needed for today's implementation.

Why?  The structure of Productions and Mu2e Offline guarantees that if the shared_ptr is valid at the start of produce it will remain valid throughout produce.  Since the deference does safety checks, we only need to pay the price of the safety check once per call to produce.  Moreover it is more expensive to pass a shared_ptr<T> as an argument than it is to pass a  T const&.  If we are playing the call and dereference overhead in a loop or nested loop it will add up.

rlcee, 2020-02-14:
Rob wrote some text on these points and then I expanded and split the ConditionsData wiki page, included the text, so it is now relatively complete and well organized.  I'll close this in a few days if no one else comments.

kutschke, 2020-02-14:
Thanks Ray,

  That looks good.

Rob

rlcee, 2020-02-17:
Wiki pages were improved..

closed 2020-02-17 by rlcee