Mu2e/Offline #98: Scrub Mu2eUtilities for inappropriate #includes

kutschke, 2020-01-09:
There are a lot of files in Mu2eUtilities that include art headers for no reason.  Remove them.

ryuwd, 2020-01-27:
I ran a tool called IWYU (https://github.com/include-what-you-use/include-what-you-use) over Mu2eUtilities. It produced these sets of proposed changes to cut out unnecessary includes (see link): 
https://gist.github.com/ryuwd/ddbcf324322b96d50aedd1b74bcb7d36

gaponenko, 2020-01-27:
Hi Ryun,  I like the spirit, but one should be careful with the implementation.   I took at quick glance at the proposed changes, and some of them seem wrong.  For example, per the C++ standard we should use the exception header.   The existence of bits/exception.h is an implementation detail of our C++ environment, and it would be wrong for us to include that header directly.

rlcee, 2020-01-27:
Years ago, we estimated that up to 30% of compile time is due to unnecessary includes.  We did not address it due to these comments:
- tool will not detect if a #define statement changed (will this tool catch that?) Since that time, we have greatly advanced our validation, so maybe this is moot?
- some includes are for clarity, future development, or commented out code.  Also my impression was that no one liked the idea of a tool editing their code.
Personally, I'm all for running this tool (module Andrei's comment)

ryuwd, 2020-01-27:
Hi Andrei, Ray,

> Hi Ryun, I like the spirit, but one should be careful with the implementation. I took at quick glance at the proposed changes, and some of them seem wrong. For example, per the C++ standard we [...]

I agree, as IWYU has some nuances (running it blindly would be bad). It might still be worth taking the output with a grain of salt, such that e.g. headers such as `<exception>` are left untouched to respect C++ standards.  At the very worst those changes/suggestions can be used just as a guide to resolve this issue

In this specific case, [a iwyu 'pragma' can be used to make sure `<exception>` is included rather than bits/exception by adding a comment to the latter file](https://github.com/include-what-you-use/include-what-you-use/blob/master/docs/IWYUPragmas.md). Or, just remove the suggested change. [I revised the gist to remove the changes to `<exception>` includes](https://gist.github.com/ryuwd/ddbcf324322b96d50aedd1b74bcb7d36/revisions).

> Years ago, we estimated that up to 30% of compile time is due to unnecessary includes. We did not address it due to these comments:
>
>     * tool will not detect if a #define statement changed (will [...]

I have used the docs as a reference:
- https://include-what-you-use.org/ (website)
- https://github.com/include-what-you-use/include-what-you-use/blob/master/README.md
- https://github.com/include-what-you-use/include-what-you-use/blob/master/docs/WhyIWYU.md
- https://github.com/include-what-you-use/include-what-you-use/blob/master/docs/WhyIWYUIsDifficult.md

IWYU sometimes produces longer include lists, as it encourages one header per symbol in use (so it is easy to add/remove dependencies without affecting others.) It removes unused header includes. It tries to replace includes with forward declares where possible. 

As far as I can tell, #defines might still be an issue (see last doc - but i imagine not so much for the extremely common header guard case.) The developers of the tool have stressed it makes mistakes occasionally (perhaps often), and that it requires corrections, but in the case of Mu2eUtilities the error rate seems relatively low.

The file I linked is a very small subset of the tool output over the entire codebase. There is a second utility that actually makes the changes, so it is possible to review the changes first.

Cheers, Ryun