Mu2e/Offline #128: Need to implement channel maps for the sub-detectors

gianipez, 2020-01-30:
We need to implement a method that assigns the proper channel-Id using as input the "hardware" coordinates given by the ROC and channel Id defined in the ArtFragment. 
This method will be used in DAQ/src/StrawAndCaloDigiFromFragment_module.cc (and in the analogous for the CRV, STM and EXTM).

rlcee, 2020-01-30:
Is this one permanent list, or does it need to be a conditions database table?

gianipez, 2020-01-31:
I had a personal conversation with the Physicist in Pisa developing the Calorimeter ROC. He told me that as long as the DTC doesn't break, the list will be permanent, which makes sense as we will not have frequent access to the ROCs inside the DS.

kutschke, 2020-01-31:
HI Giani,

   What happens if a card fails and you swap in a replacement?  Or you do maintenance and swap two cards when you put things back?

Rob

gianipez, 2020-01-31:
Hi Rob,

By "card" you mean the DTC? From what I remember, we are not going to have a large number of spares for the DTCs and FPGA used to read out the optical links, so my guess is that the probability for that to happen will be limited. I'll ask Ryan the frequency he expects this operation to happen. 

Giani

kutschke, 2020-01-31:
Hi Giani,

  I don't care about the probability of it happening.  If it can happen at all then this information should be in the conditions data base not hard coded.

    Rob

rlcee, 2020-01-31:
Why is the readout chain (I'm not sure where/how these numbers are generated) 
reporting physical location and not a programmable function number?
In other words, why isn't a particular readout path/art fragment producing a
programmable ID that says "I'm attached to the north side of disk 2" 
and not "I'm sitting in rack 2, slot 3" or "I'm ROC board serial number N"?
I don't know anything about these boards and their configuration, so maybe
there is a good reason.

gianipez, 2020-01-31:
Hi Ray, all these info are present in the data-fragment sent by the ROC and read out by the DTC. What is missing is then the assignment to a unique Id. For example: "ROC 0, channel 10 is calo-readout-id 120"...

rlcee, 2023-07-17:
@gianipez @kutschke I believe this functionality is now in the CaloDAQMap proditions.  Can we close this Issue?  Thanks

kutschke, 2023-07-18:
Is the equivalent functionality in place for all other detectors?  If not, which are missing?

rlcee, 2023-07-18:
I think this is in place for tracker, calo, crv, and stm.  So only EM missing.  I have heard no reqs or plans for that.

kutschke, 2023-07-18:
> I think this is in place for tracker, calo, crv, and stm. So only EM missing. I have heard no reqs or plans for that.

Ok.  Then it's Ok by me to close the issue.  When Andrei starts to work on the ExtMon software again we can ask him about that.

rlcee, 2023-07-18:
This Issue was from @gianipez , so Giani, can we close it?  Thanks

closed 2023-07-20 by gianipez