Mu2e/Offline #133: Calorimeter template waveform is deprecated and about 4 times shorter than the real one.

gianipez, 2020-02-04:
We need to update the template used to generate the simulated SiPM response in the calorimeter simulation. The current waveform is about 4 times shorter than the one obtained with the latest version of the readout electronics.

brownd1978, 2020-02-04:
What downstream changes need to go with this?  Does the digi format
change?  Will calibrations need updating?  Would this be a good opportunity
to migrate the calorimeter calibrations to Proditions?

gianipez, 2020-02-04:
The required downstream changes are quite a few:
- different calibration constant(s)
- different function for time extraction

Sophie will discuss at [this meeting](https://mu2e-docdb.fnal.gov/cgi-bin/private/DisplayMeeting?sessionid=8709) the new digi format that reflects the current firmware output. We can probably address the Proditions question at that meeting.

Even if by changing the waveform we still have no realistic model of the timing resolution, I would like to do it anyway because it will improve our understanding of the online calo-clustering performance.  In the online reco, we are currently using the peak positions of the waveforms found by the DIRAC. With longer pulses, I expect an increase in the pileup that will deteriorate the performance.

kutschke, 2021-04-16:
Bertrand reports that this has been addressed by recent PRs.

closed 2021-04-16 by kutschke