Mu2e/Offline #155: const correctness in BFieldManager ?

kutschke, 2020-03-11:
Something to check:

Consider the following fragment from BFieldGeom/inc/BFieldManager.hh:

class BFieldManager {
public:
   typedef std::vector<std::shared_ptr<BFMap>> MapContainerType;
   const MapContainerType& getInnerMaps() const { return innerMaps_; } 
private:
    MapContainerType innerMaps_;
 };

I *think* that the const accessor getInnerMaps() actually gives non-const access to the Maps!   The const only protects the vector itself, not the BFMap objects to which the entries point.

I am not sure the easiest way to fix this.  Can we do something like:
   typedef std::vector<std::shared_ptr<const BFMap>> MapContainerType;

Does that cause problems creating the BFMap objects?  If so we can make a transient object of type std::vector<std::shared_ptr<BFMap>> and then copy the final result to the data member?

Are there similar issues in the proditions system which also makes use of shared_ptr?

ryuwd, 2020-03-11:
Hi Rob,
(I can't read your C++ types correctly, I think github doesn't like unescaped angle brackets)
I believe you're right that the BFMap objects are not const in a const vector of `shared_ptr<BFMap>`, but `shared_ptr<BFMap const>` would provide the desired const behaviour.

I don't think you can't cast `std::vector<shared_ptr<T>>` to `std::vector<shared_ptr<T const>>` directly but you could maybe make a copy of the vector by `const_pointer_cast<BFMap const>(<each element>);`  on each element? you'd have double the number of shared_ptr's but the same number of BFMaps in memory

I think ProditionsHandle prevents this kind of problem by holding and providing `shared_ptr<T const>`, doing a `dynamic_pointer_cast` when it needs to initially get a const ptr given a non-const pointer from the cache.
https://github.com/Mu2e/Offline/blob/9e5e9f00ec9a213ec5079375a0e666d22339294c/ProditionsService/inc/ProditionsHandle.hh#L39-L50

http://www.cplusplus.com/reference/memory/const_pointer_cast/
http://www.cplusplus.com/reference/memory/dynamic_pointer_cast/

rlcee, 2020-03-11:
Ryun, you beat me to it - I was about to say the same.  All of Proditions uses
    ```typedef std::shared_ptr<const mu2e::DbTable> cptr_t;```
except in a few considered places.  The shared pointers and vectors of shared pointers are not const.

kutschke, 2020-03-11:
Thanks !  So the issue is limited to BFieldManager.  That's great news.

kutschke, 2020-03-11:
> Hi Rob,
> (I can't read your C++ types correctly, I think github doesn't like unescaped angle brackets)
> I believe you're right that the BFMap objects are not const in a const vector of [...]

We would need the non-const version in memory only transiently while the BFMap objects are created - once the initialization phase is over, the non-const version can go away.  The memory use is small enough that this is OK.

rlcee, 2022-10-12:
Fields now held with shared_prtr<const BFMap>

closed 2022-10-12 by rlcee