Mu2e/Offline #230: NaN's as a way of return status in CRV code

kutschke, 2020-08-20:
At the Wed Aug 19, 2020 weekly software and computing meeting Ralf reported that when a CRV reco pulse fails the fit he used to discard it.  In the interest of improving efficiency he now keeps the pulse but sets some values to NaN; other values he can set to the value used as input to the fit (determined by an earlier step).

We request that he add a status word that describes the status of the reco pulse and never set a value to NaN; instead set those values to "safe" values.    Presumuably there might be many classifications of pulses so don't use a bool to flag good/bad; instead use something like a bitmap to to hold the status.  See, for example, RecoDataProducts/inc/StrawHitFlag.hh .

kutschke, 2021-04-15:
Today the string isnan appears 63 times in our .cc files.  It does not appear in our hh files.

rlcee, 2022-08-24:
Ralf believes he has fixed the main issue here, the reco pulse product.  There are still 62 NaN's in use (some new ones added), but they appear to be local use only, or catching real numeric error conditions, though I only did a good sampling, and did not study each use.  Is this good enough or do you want each use examined, and or removed?  If a full accounting is needed, what criteria makes it a problem?

rlcee, 2023-07-19:
Major progress in #1014.  @ehrlich-uva Another major hot spot is Offline/EventDisplay.  Do you want to take a look at that?  Or I could.  Also at times we have talked about separating EventDisplay from Offline and making a new repo.  Let me know if you are OK with that, I could do it.  Thanks

rlcee, 2023-07-21:
> Major progress in #1014. @ehrlich-uva Another major hot spot is Offline/EventDisplay. Do you want to take a look at that? Or I could. Also at times we have talked about separating EventDisplay from Offline and making a new repo. Let me know if you are OK with that, I could do it. Thanks

@ehrlich-uva I hope you can take a minute to comment on this question.  The end of this issue is in sight.

ehrlich-uva, 2023-07-21:
Is it necessary to remove the NANs from the EventDisplay? They are only used locally inside the module and are not passed to other modules or written to files.

rlcee, 2023-07-21:
> Is it necessary to remove the NANs from the EventDisplay? They are only used locally inside the module and are not passed to other modules or written to files.

This is Rob's request so I'll leave that answer for him.  I'm happy to do the editing - my goal is to close Issues (reasonably).  How do you feel about splitting EventDisplay into its own repo?  I think we agreed to do that at some point.

ehrlich-uva, 2023-07-22:
I can work on these changes.

rlcee, 2023-08-14:
An analysis of what it will take to close this issue. Assume we will eliminate std::quiet_NaN as well as NAN.  Ralf, I'm happy to do the work.  @kutschke @ehrlich-uva @gaponenko

NaN is used in a few ways:
A) initialize variables
   propose replace with std::numerical_limits<DBL_MAX> (or infinity?)
B) as a flag for something not done
    add bools
C) a check on calculations that when wrong
    leave these in

> egrep -r -l "NAN|NaN|isnan" *

`Analyses/src/CosmicAnalysis_module.cc`
type A,B
```
Analyses/src/EMFCSimpleDumper_module.cc
Analyses/src/SimParticleDumperCompact_module.cc
Analyses/src/StepPointMC1stHitDumper_module.cc
Analyses/src/StepPointMCDumperCompact_module.cc
Analyses/src/StepPointMCDumper_module.cc
Analyses/src/SensitiveTargetEnergyDumper_module.cc
Analyses/src/PrimaryProtonEnergyDumper_module.cc
```
type A

```
CosmicReco/src/CosmicFitDisplay_module.cc
CosmicReco/src/CosmicTrackFit.cc
CosmicReco/src/MinuitDriftFitter.cc
```
type C

`CRVAnalysis/src/CRVAnalysis.cc`
type A
`CRVResponse/src/CrvPhotonGenerator_module.cc`
type C
`CRVResponse/src/CrvPlot_module.cc`
type A,B,C
`CRVResponse/src/DrawLookupTableHistograms.cc`
type C
`EventDisplay/src/*`
move to its own repo, make Muse compliant

`GeneralUtilities/src/Binning.cc`
return DBL_MAX

```
MCDataProducts/inc/CrvPhotons.hh
MCDataProducts/inc/CrvSiPMCharges.hh
```
type A, I thought these were already removed?  

`Mu2eKinKal/src/KKStrawMaterial.cc`
type C

```
Mu2eUtilities/inc/BinnedSpectrum.hh
Mu2eUtilities/inc/polyAtan2.hh
```
in comments, ignore

ehrlich-uva, 2023-08-14:
I will work on my projects

rlcee, 2023-08-16:
Seeing no comment, I will proceed with the non-CRV proposed work.