Mu2e/Offline #235: Hardcoded physics values in Mu2eUtilities/src/BinnedSpectrum.cc

gaponenko, 2020-08-26:
The code  in Mu2eUtilities/src/BinnedSpectrum.cc contains things like

      const double bindingEnergyFit{0.464};
      const double recoilEnergyFit {0.220};
      const double deltaMassFit    {3.121};

which are wrong for non-aluminum targets, or

      double ehi = 105.; // cut off at muon mass

brownd1978, 2020-08-26:
Please make this an issue tied to Mu2eUtilities.  It's not directly tied to
this PR.
Dave

brownd1978, 2020-08-26:
Opps, my bad, it is an issue!  I misread the #235 as #234

Dave

kutschke, 2021-04-15:
Following up on this.  There are 22 occurrences of the string 3.14 in our code - not all are values of PI.  Some are in tables of physical properties.

rlcee, 2022-08-24:
Should these three values just go into globalconstants?  I assume muon mass should come from global constants.  I'll make a separate PR for pi.

rlcee, 2022-10-17:
Hardcoded physics values fixed in #857

closed 2022-10-17 by rlcee