Mu2e/Offline #292: Synchronizing pdgcodes across Offline, BTrk and KinKal is a bigger issue.  Fortunately they all use the same values.  It's clear though this code should use PDGCode values, not hardcoded numbers.

kutschke, 2020-12-10:
Synchronizing pdgcodes across Offline, BTrk and KinKal is a bigger issue.  Fortunately they all use the same values.  It's clear though this code should use PDGCode values, not hardcoded numbers.

_Originally posted by @brownd1978 in https://github.com/Mu2e/Offline/pull/286#discussion_r540434072_

kutschke, 2020-12-10:
A cursory inspection of the code base found about 50 places that contain hard coded PDG codes, where the PDGCode enum names should have been used.

brownd1978, 2020-12-10:
Hi Rob,

> A cursory inspection of the code base found about 50 places that contain
> hard coded PDG codes, where the PDGCode enum names should have been used.

Were these all .cc code or also .fcl?  That latter is also a problem, with
no easy solution.  I did centralize the numbers for TrkPatRec in a fhicl
namespace, we should probably move that to a more general directory.

Dave

> <https://github.com/Mu2e/Offline/issues/292#issuecomment-742806379>, or
> unsubscribe
-- 
David Nathan Brown            Dave_Brown@lbl.gov
Office Phone (510) 486-7261  Fax  495-2957
Lawrence Berkeley National Lab
MS 50R5008 (50-6026C)  Berkeley, CA 94720

kutschke, 2020-12-10:
Hi Dave,

  I just grepped for the strings 211 and -11 in all .cc files and filtered out a few obvious false positives.  I did not look at fcl.

  A candidate solution for fcl is to define something like  Offline/fcl/enums.fcl

BEGIN_PROLOG
PDGCodes : {
  eminus   : 11
  eplus    : -11
 mu_minus : 13
 mu_plus  : -13
}
END_PROLOG

And so on.  We should spell the names the same as for PDGCode.hh .  Then #include it in the appropriate spots (maybe some thinking here?).
We could define other “enums” in that fcl file as well.

Rob

On Thu, Dec 10, 2020 at 13:20 Rob Kutschke <notifications@github.com> wrote:

> A cursory inspection of the code base found about 50 places that contain
> hard coded PDG codes, where the PDGCode enum names should have been used.
>

Were these all .cc code or also .fcl? That latter is also a problem, with
no easy solution. I did centralize the numbers for TrkPatRec in a fhicl
namespace, we should probably move that to a more general directory.

.

rlcee, 2022-09-26:
@brownd1978 it look like there are no PDG numbers in KinKal, is that right?
@kutschke it looks like #462 is a duplicate of this conversation.  Without further comment, I'll close that Issue and fhicl can be addressed on this issue

brownd1978, 2022-09-26:
Hi Ray,

> @brownd1978 <https://github.com/brownd1978> it look like there are no PDG
> numbers in KinKal, is that right?
Sadly there are lots: see
https://github.com/Mu2e/Offline/blob/6800e132f2ac1f0657b85d6dc60e0b7c8bac607c/Mu2eKinKal/fcl/prolog.fcl#L245
and below.   All of these should use names, not raw numbers.  Also upstream
and downstream should be names, not numbers.
There are several ways to accomplish this.  One is common fcl to define
these numbers, another is to define an enum class with a text interface to
convert strings to enum values.

> @kutschke <https://github.com/kutschke> it looks like #462
> <https://github.com/Mu2e/Offline/issues/462> is a duplicate of this
> conversation. Without further comment, I'll close that Issue and fhicl can
> be addressed on this issue
>
> <https://github.com/Mu2e/Offline/issues/292#issuecomment-1258619385>, or
> unsubscribe

rlcee, 2022-09-26:
> Sadly there are lots:
I wasn't clear.  I meant it looks like there are no explicit PDG codes as numbers in the KinKal repo or product.  There are many occurrences in fhicl which I hope to address all at once.  For now, I'm confirming that there is only fhicl left to do - code is done.

kutschke, 2022-09-27:
I want to put on the record two options we discussed during a meeting a few weeks ago.

Option 1:
Start from    https://github.com/Mu2e/Offline/blob/main/TrkPatRec/fcl/Particle.fcl and rename it to maybe Offline/fcl  so that it's clear that it's of general use.

Option 2:
Use text string names in fcl, maybe "electron", or "e-" or "e_minus" ...  
Add support in ParticleDataList for these names.  Add a utiltiy to transform a fhicl sequence of these names to a std::vector or std::array of PDGCode objects

The second option is more work up front but more user friendly in the long run and I prefer it.

brownd1978, 2022-09-27:
> I want to put on the record two options we discussed during a meeting a
> few weeks ago.
>
> Option 1:
> Start from
> https://github.com/Mu2e/Offline/blob/main/TrkPatRec/fcl/Particle.fcl and
> rename it to [...]
The big benefit of option 2 is that the debug-config output will list human
readable names, not numbers.
I strongly support option 2, and if possible, suggest making it a generic
(templated) utility for assigning enum values based on text strings.  We
need the same technology for
configuring upstream and downstream fits, and there are plenty of other
similar cases, there's no point in writing that code again and again.
I hand-coded this in Mu2eKinKal to allow configuring the updater algorithms
by name, take a look at
https://github.com/Mu2e/Offline/blob/main/Mu2eKinKal/src/StrawHitUpdaters.cc

> The second option is more work up front but more user friendly in the long
> run and I prefer it.
>
> <https://github.com/Mu2e/Offline/issues/292#issuecomment-1259731212>, or
> unsubscribe