Mu2e/Offline #415: Remove temporary code to allow backwards compatibility with older files of CRV event data

kutschke, 2021-04-21:
There is currently one temporary fix in the CRV code to allow the head of Offline/master to work with files containing older CRV event data.  There will soon be a new one.  Once the older files are declared obsolete, these fixes should be removed.

The first one deals with special treatment of older StepPointMCs that are missing a data member.

The second one deals with the changes to the counter numbering caused by removal of the counters around the cryo window.

rlcee, 2022-10-14:
@kutschke @ehrlich-uva what is the status of this Issue?  It is 18mo old now. Thanks

ehrlich-uva, 2022-10-14:
I think the first point refers to CrvStepsFromStepPointMCs_module. Yuri added options to deal with older StepPointMCs that didn't have the visibleEDep and postPosition. I don't know whether this is still needed.
The second point is about the current geometry that contains some CRV modules around the cryo hole that are not part of the design anymore. I could make a PR for the new geometry, and people who want to read older files could use the previous geometry.
I will discuss both points with Yuri @oksuzian

ehrlich-uva, 2022-10-26:
I can remove the temporary code in CrvStepsFromStepPointMCs_module. However, we need to keep the CRV geometry until we start the next big simulation.

rlcee, 2026-01-30:
@ehrlich-uva please comment on the status of this issue, thanks

ehrlich-uva, 2026-01-30:
The new CRV geometries are in place now (for the full geometry, the KPP, and Run1a). The older geometry still exists for MDC2020.
@oksuzian Do you know, if we still have older files where the StepPointMCs don't have visibleEDep and postPosition? If not, we can remove these options that deal with these situations.