Mu2e/Offline #426: Offline JobConfig dependence

rlcee, 2021-04-27:
JobConfig will be split off to its own repo.  Here is a summary of Offline dependence on JobConfig.  All these will need to be dealt with in order to do the split cleanly.  Or maybe each can be fixed when needed?

********** using JobConfig/common/geom_baseline.txt
Analyses/fcl/PionMomentumAnalyzer.fcl
Analyses/fcl/pbars1hist.fcl
Mu2eG4/fcl/iontest_g4s2.fcl
EventGenerator/fcl/cryTest.fcl
ExtinctionMonitorFNAL/test/emfRecoTrackAnalysis.fcl
Mu2eG4/geom/studies/geom_cosmic_v05.txt

********** non-trivial - move to Production?
   using  #include "JobConfig/reco/Reco.fcl"
TrkDiag/fcl/TrkAnaDigisReco.fcl

  using #include "JobConfig/reco/prolog.fcl"
Mu2eKinKal/fcl/SeedTest.fcl

********** event displays - move to Production?
  using #include "JobConfig/reco/mcdigis_CRY.fcl"
EventDisplay/fcl/EventDisplayDigis.fcl

  using #include "JobConfig/reco/mcdigis_nocosmic.fcl"
TEveEventDisplay/CallerFclExamples/cosmicExample.fcl
TEveEventDisplay/CallerFclExamples/cosmictracks.fcl

  using #include "JobConfig/beam/prolog.fcl"
TEveEventDisplay/src/oldprolog.fcl

**********   this will be split off in its own repo, 
       maybe move this fcl to that repo
  using #include "JobConfig/alignment/collect_nofield_cosmictracks.fcl"
           #include "JobConfig/reco/misalign_epilog.fcl"
TrackerAlignment/fcl/job_template.fcl

********** in test, expert can fix
Analyses/test/Mu2eCCFC_conversion.fcl
Analyses/test/Mu2eCCFC_dio.fcl
Analyses/test/genMixReco.fcl
Analyses/test/readTrackCluster.fcl
Analyses/test/vd_readStage2.fcl
EventGenerator/test/ceLeadingLog_test.fcl
EventGenerator/test/ceMEndpoint_test.fcl
EventGenerator/test/ceMLeadingLog_test.fcl
EventGenerator/test/cePEndpoint_test.fcl
EventGenerator/test/cePLeadingLog_test.fcl
EventGenerator/test/corsikaTest.fcl
EventGenerator/test/testStoppedParticleReactionGun.fcl
TrkPatRec/test/TrkPatRecTestMix.fcl

********** will move to Production
CampaignConfig/corsika.cfg
CampaignConfig/mdc2020.cfg
Validation/fcl/ceMixDigi.fcl
Validation/fcl/ceSimReco.fcl
Validation/fcl/cosmicSimReco.fcl
Validation/fcl/reco.fcl
Validation/test/ceDigi.fcl
Validation/test/ceMix.fcl
Validation/test/ceSteps.fcl
Validation/test/muSteps.fcl

********** not an include 
EventMixing/fcl/prolog.fcl

brownd1978, 2021-04-28:
Hi Ray,

> JobConfig will be split off to its own repo. Here is a summary of Offline
> dependence on JobConfig. All these will need to be dealt with in order to
> do the split cleanly. Or maybe each can be fixed [...]
I would move the fcl files to /test.  The geometry file looks circular.  I
think  we should consolidate all the geometry, either in Mu2eG4/geom (my
preference) or  JobConfig/common.  Dave (Louisville) and Yuri, what is your
opinion?

> ********** non-trivial - move to Production?
> using #include "JobConfig/reco/Reco.fcl"
> TrkDiag/fcl/TrkAnaDigisReco.fcl
This will move with TrackAna to its new repo.

> using #include "JobConfig/reco/prolog.fcl"
> Mu2eKinKal/fcl/SeedTest.fcl
This can go into /test

> ********** event displays - move to Production?
> using #include "JobConfig/reco/mcdigis_CRY.fcl"
> EventDisplay/fcl/EventDisplayDigis.fcl
>
> using #include "JobConfig/reco/mcdigis_nocosmic.fcl" [...]
I'm not sure we ever concluded, but the consensus seemed to be that the
EventDisplays should be in their own repos, in which case these scripts
would move with them.

> ********** this will be split off in its own repo,
> maybe move this fcl to that repo
> using #include "JobConfig/alignment/collect_nofield_cosmictracks.fcl"
> #include "JobConfig/reco/misalign_epilog.fcl"
> TrackerAlignment/fcl/job_template.fcl
misalign_epilog is specific to a campaign.  It needs to stay in JobConfig.

I think JobConfig/alignment should move to the TrackerAlignment repo, but
that's really for Richie etal to decide.

> ********** in test, expert can fix
> Analyses/test/Mu2eCCFC_conversion.fcl
> Analyses/test/Mu2eCCFC_dio.fcl
> Analyses/test/genMixReco.fcl
> Analyses/test/readTrackCluster.fcl
> Analyses/test/vd_readStage2.fcl
Analyses should be its own repo

> EventGenerator/test/ceLeadingLog_test.fcl
> EventGenerator/test/ceMEndpoint_test.fcl
> EventGenerator/test/ceMLeadingLog_test.fcl
> EventGenerator/test/cePEndpoint_test.fcl [...]
These could be truely circular as JobConfig depends on EventGenerator.  I
think they should all be moved to /test

> TrkPatRec/test/TrkPatRecTestMix.fcl
this file should be removed.

> ********** will move to Production
> CampaignConfig/corsika.cfg
> CampaignConfig/mdc2020.cfg
> Validation/fcl/ceMixDigi.fcl
> Validation/fcl/ceSimReco.fcl
> Validation/fcl/cosmicSimReco.fcl [...]
Having multiple validation directories in Production would be confusing.
How do you plan to consolidate JobConfig/validation and Validation/fcl and
Valdation/test?

> ********** not an include
> EventMixing/fcl/prolog.fcl
This file is empty: not sure why it is here.

rlcee, 2021-04-29:
> How do you plan to consolidate JobConfig/validation and Validation/fcl and Valdation/test?

I'd put them all under Production/Validation/fcl/..  I suggest fcl instead of test since this fcl is the primary purpose of this repo, not a test.  But I don't feel strongly about the details.  

If no one else comments by tomorrow, I'll implement Dave's plan.  Directories to be split to other repos I'll leave in place and let them break (when JobConfig is moved).

brownd1978, 2022-08-25:
Hi Ray,

> Assuming it is OK to have a fcl dependence loop in a test directory, there
> are the following remaining issues and suggested fixes.
>
> CaloMC/fcl/RunCaloCalibGun.fcl
>
> move this to CaloMC/test
>
> Mu2eKinKal/fcl/KKSeed.fcl
> Mu2eKinKal/fcl/KKDrift.fcl
>
> move these to Mu2eKinKal/test

I'd rather do this in PR 828.  That way there will be fewer merge conflicts.
cheers, Dave

> Sources/test/FromTrackerPrototype.fcl
>
> since this is not test fcl, it reads Tracker VST bin data and makes an art
> file, maybe best is to move it to Production
>
> TEveEventDisplay/CallerFclExamples/POT.fcl
>
> this doesn't seem appropriate to TEveEventDisplay and is broken. Suggest
> deleting it
>
> I will do these actions in PR's on Monday if there is no objection
>
> <https://github.com/Mu2e/Offline/issues/426#issuecomment-1227321350>, or
> unsubscribe

rlcee, 2022-09-13:
with merge of #828, this is complete

closed 2022-09-13 by rlcee