Mu2e/Offline #482: Investigate event reproducibility in jobs using Mu2eG4MT_module

resnegfk, 2021-06-09:


kutschke, 2022-03-26:
Krzysztof reported that the reproducer is:

> mu2e -n 250000 -c Production/JobConfig/validation/stoppedMuonsSingleStageMT.fcl >& stoppedMuonsSingleStageMT.log
> This takes about 40min.

This runs 5 threads and 5 schedules.  Two consecutive runs of this job produce a different number of output events.  I added some printout to see which events were being written out and discovered that there is very little overlap in the events that are written out, maybe 10%.

The issue is not in G4MT but in the interaction of the event generator EventGenerator/src/PrimaryProtonGun_module.cc with the MT environment.  When you run the same job 2 times,  the match of event number to the instance of PrimaryProtonGun is random - it depends on the thread scheduler and environmental effects.  So in two separate jobs the match of generated event to G4 worker threads is broken.

This is not a problem for physics purposes but it is a problem because we do not have the repeatability needed for debugging purposes.  Issue #745 has been created to address this.

This investigation also uncovered an unrelated problem in EventGenerator/src/PrimaryProtonGun_module.cc:  all instances get the same seed, which is reported in issue  #744 .

kutschke, 2022-03-29:
The immediate problems have been fixed with PR #747.  The new issues #744 and #745 have been created to create a better solution.

closed 2022-03-29 by kutschke