Mu2e/Offline #484: Bad Physi8csParams defaults

gaponenko, 2021-06-10:
When GlobalConstantsService loads its configuration from a SimpleConfig file, most parameters are required to be explicitly defined.  But there are a few dangerous lines in the code like

    _captureProtonRate[material] = config.getDouble("physicsParams."+material+".capture.protonRate", 0);
    _captureDeuteronRate[material] = config.getDouble("physicsParams."+material+".capture.deuteronRate", 0);
    _captureNeutronRate[material] = config.getDouble("physicsParams."+material+".capture.neutronRate", 0);
    _capturePhotonRate[material] = config.getDouble("physicsParams."+material+".capture.photonRate", 0);

which default undefined physics rates to zero, potentially leading to jobs happily producing wrong physics output.
I am going to add an explicit exception  throw for non-aluminum material to Pileup_module.cc until this is fixed.

rlcee, 2022-09-13:
Some information is provided for all materials, and some is not.
I think there are three options here
- do not load anything for a material unless all numbers are available (shorten material list)
- provide all needed numbers for all materials (need a reference)
- load available numbers (create two materials lists for loading purposes, complete materials and incomplete materials)
@kutschke @gaponenko

gaponenko, 2022-10-11:
I think we ned to allow materials with incomplete information to be used.   For example, if someone studies Mu2e-II physics sensitivity in the mu- to e- channel, they do not need the mu- to e+ channel energy, and if the latter is missing it should not prevent the study in the "information available" channel.   I don't think there should be two lists of materials.  All materials can be treated uniformly, and each accessor must either return valid data or throw.

rlcee, 2022-10-12:
Here are some ways to implement this, @gaponenko please comment
- attempt to read all values for all materials, use default value of DBL_MAX, then check for this value in the getters.
- call hasName to check the file for each value, keep track of what is available in a PhysicsParams bit flag, check this in the getters
- other?

The first involves a little less code, but uses a flag value in the physics variable.

gaponenko, 2022-10-12:
Keeping track of what is available and storing values as
std::optional<double> rather than double would be a clean solution,
but the DBL_MAX flag value approach is probably a lot simpler to
implement and looks good enough to me.   I am fine with either solution.

Andrei

rlcee, 2022-10-12:
One more possible model.  Save the available values as `map<string,double>`, change
the accessors to `getValue(string targetMaterial, string value)`.  This would compress the code a lot at the cost of doing a second short string search.  If this code is called only in initialization routines, then no big deal.  @gaponenko please comment (I'd like to do this only once..)

gaponenko, 2022-10-12:
That feels too loose to me.  String targetMaterial is OK, and that's
what we've been using, but I don't like the open-ended "string value".
My preference is to keep all parameters enumerated in the API, like
getEndpointEnergy(material)

Actually the current PhysicsParams implementation already uses maps
indexed by material name, so missing parameters are intrinsically
allowed.   It is the filling and accessing that need to be changed.
And adding any new types of parameters (the RMC ones).

Andrei

rlcee, 2022-10-17:
Addressed in #857

closed 2022-10-17 by rlcee