Mu2e/Offline #500: Use of .at(i) vs operator[](i)

kutschke, 2021-06-20:
I was looking at some code that is in a time-critical algorithm that runs in the trigger.  It used .at(i) to access elements of a std::vector; the index i is computed in a way that it is safe to use [i].   It's still a good idea to use .at(i) for debugging and testing, in order to catch errors in the surrounding code.

If this use of at(i) is just in this one place it's not likely to be a significant time issue and we can let it ride.  If it's ubiquitous, we should adopt one of the methods of enabling/disabling run-time bounds checking with a compile time switch.

The case that I noticed is in PR #499 in the file CaloReco/src/CaloHitMakerFast_module.cc  .  There is a comment in the review at line 113 of the incoming file.

It may be useful to think about this when dealing with issue #439  .

rlcee, 2022-10-10:
How can I see if it's "ubiquitous"? What might be a method to make a list of cc files related to the trigger?  Listing the modules should be straightforward, but what significant code are they calling? Or just focus on a set of subdirectories?

We believe we can catch vector overrun problems with the gnu sanitize check.  We run this test on reco every week, @kutschke is that enough?  I can run other examples now, if needed.  I'd expect to run thorough tests like this as we get closer to actually running the trigger in production.

brownd1978, 2022-10-10:
I would ignore use of at() outside of nested event loop processing.  I use
it in init methods where it provides fast trapping of configuration errors.
I t should be fairly easy to decide which cases apply.

kutschke, 2022-10-24:
I generally do the same as Dave - in initialization code and other run-rarely code I use at().  And I use [] in loops in per event code.  Should we propose this as a best practice we ask people to adopt?   In the very early days of Mu2e I was more paranoid and some of that code survives.

I believe that gnu sanitize is good enough; the open question is how often to do it and for how many events.  What we are doing now is good enough for now.   As we develop policies (and people to do the work) to bless new releases, we should decide on how much sanitization is appropriate.  And we should ask Krzysztof about what G4 does for testing their code.