Mu2e/Offline #530: Loop dependencies in fcl files

kutschke, 2021-07-15:
The following .fcl files in Offline contain references to Production/

./Analyses/test/Mu2eCCFC_conversion.fcl
./Analyses/test/Mu2eCCFC_dio.fcl
./Mu2eKinKal/test/SeedTest.fcl
./TEveEventDisplay/CallerFclExamples/POT.fcl
./Validation/test/ceDigi.fcl
./Validation/test/ceMix.fcl
./Validation/test/ceSteps.fcl
./Validation/test/muDauSteps.fcl

We need to understand where to move these to undo the loop dependency.

rlcee, 2021-07-15:
The Validation files are obsolete, those were moved to Production.  I'll delete them

kutschke, 2021-07-15:
> The Validation files are obsolete, those were moved to Production. I'll delete them

Thanks Ray.  I will talk with Sophie about TEveEventDisplay and with Bob about the two in Analysis.

Dave: can look at the file in Mu2eKinKal.

rlcee, 2021-07-15:
Validation files removed in #531

brownd1978, 2021-07-16:
I thought we agreed to not consider /test contents as subject to this loop
test.  The KinKal dependence is not a real (circular) loop.

If people are really bothered by this the /test content can be migrated to
Production/Tests.  I don't think it solves any actual problems though.
Dave

kutschke, 2021-07-16:
After the email that you saw I added a comment about test directories.  I am Ok sticking with that policy for cases where it's  thoughtfully considered and it's the right place for it.  It sounds like the KinKal case satsifies this.

rlcee, 2021-07-17:
I would vote to organize the dependencies logically, i.e. no reference to Production in Offline.  It's just a bad example.

brownd1978, 2021-07-17:
I propose to create a Tests repository, which can reference both Offline
and Production.  I would make it open to anyone who wants to commit to it,
with no 'code review'.  We should use it only
for non-essential test code.  It will keep us from having to update Offline
everytime some thinks of a new test plot.  I would move everything in
TrkDiag/tests into Tests/Tracking.  What do you think?

Dave

rlcee, 2021-07-17:
I think creating new repos that depend on Production and Offline is the logical conclusion of 
- they have these dependencies
- they don't really belong in Production
I'd only pause to think about what other fcl might belong here, like Production/Validation, JobConfig/test or validation.   Should it then be called Validation?  There are more examples of this category coming, like JobConfig/alignment, event displays, calibration or ntuple making, etc.  Really anything that might need to be aware of the reco path, or might need to generate sim samples.

Another observation: the current dependency order is
* Tutorial TrackerAlignment TrkAna Production Offline
So really Production/alignment could go back in TrackerAlignment, which feels very right.  Event displays and ntuple makers could be handled the same way.

brownd1978, 2021-07-19:
I think the preconditions for 'Test' preclude any overlap with Production (
Validation or JobConfig).   Ff a 'test' is really used to generate a
prodition (say) then it
belongs in the Production repo.

I think we've converged on the concept of Tests repository.    I propose to
structure it as Tests/Tracking, etc, and  content from TrkDiag and
Mu2eKinKal there now.
Do we agree?  If so I'll create it today.

kutschke, 2021-07-19:
Hi Dave,

  I am not 100% certain that we will end up with this but let’s start and see where it takes us.

Rob

rlcee, 2021-07-19:
>I think the preconditions for 'Test' preclude any overlap with Production ( Validation or JobConfig). 

Are you referring to the suggestion that Production/Validation be moved to test?  Why not move it? Tests is a better name for a repo of validation than Production.  But I don't feel strongly. 

>If a 'test' is really used to generate a prodition (say) then it belongs in the Production repo. 

Are you referring to my suggestion to move Production/JobConfig/alignment back to TrackerConditions?  It sounds like you are proposing a rule that all Pass1 fcl is in Production.  That sounds reasonable, but it is actually still breaking the dependency order.  That fcl in Production does refer to prologs and modules in TrackerAlignment, which is above Production in the dependency order, making a loop.  You could imagine putting TrackerAlignment below Production in the dependency order, but then everyone who uses Production has to be aware of TrackerAlignment.

As far as I am concerned, you can create and populate Test.  I don't feel strongly where Validation ends up.  I do feel strongly about the TrackerAlignment issue.

brownd1978, 2021-07-19:
> I think the preconditions for 'Test' preclude any overlap with Production
> ( Validation or JobConfig).
>
> Are you referring to the suggestion that Production/Validation be moved to
> test? Why not move it? Tests is a better name for a repo of validation than
> Production. But I don't feel strongly.
I agree it is more logical to have Validation in Tests.  If you'd like to
move it after Tests is created thats fine with me.

> If a 'test' is really used to generate a prodition (say) then it belongs
> in the Production repo.
>
> Are you referring to my suggestion to move Production/JobConfig/alignment
> back to TrackerConditions? [...]
Yes, my model is Production produces all the art files needed for all
downstream consumers, including alignment.
The 2 scripts currently in TrackerAlignment that source Production fcl
should be split, so that Production/reco produces the art files, then the
TrackerAlignment jobs run on those
to produce the Millipede files. That will break the dependency.
I think the split should be easy, the OffSpill and Extracted triggers
includes lines for alignment, and they can be used to create an appropriate
reco stream. I'll work with Richie on it.

There is also
#include "Production/JobConfig/reco/misalign_epilog.fcl"
This is a campaign-specific configuration file whose content should come
from Proditions.  So it (and its sub-files) should be removed.  Is that
what you're referring to above (moving back to TrackerConditions)?

> As far as I am concerned, you can create and populate Test. I don't feel
> strongly where Validation ends up. I do feel strongly about the
> TrackerAlignment issue.
The main point of Tests is to avoid tying Offline builds to updates in QC
codes.   I think the long-term evolution should be to move everything
currently in Offline/Xxx/tests into Tests/Xxx.   I'll propose this on
Wednesday
in the meeting.

Dave

> <https://github.com/Mu2e/Offline/issues/530#issuecomment-882840963>, or
> unsubscribe

rlcee, 2021-07-20:
If I understand, your model is that TrackerAlignment and Production are peers in the dependency tree, that neither depends on the other.  Production runs reco and produces a file, then TrackerAlignment code and fcl makes ntuples. This strategy which you might call "isolation by data file" is one I favor.  I use it in the nightly validation jobs and I suggested it to separate EventDisplay from reco.

Some current fcl in TrackerAlignment refers to Production and some fcl in Production/JobConfig/alignment refers to TrackerAlignment.  I'm not sure what each does, but both need to be dealt with in this scheme, and I assume it can be done.

In the scheme I was proposing, TrackerAlignment is above Production in the dependency order and therefore may depend on Production.  The ntuple-maker can obviously live here, but optionally, if the alignment needs to modify reco, by turning off a module, or employing a module in TrackerAlignment (maybe a filter), or whatever, then that specialized reco can also live here. 

Enforcing peer repos limits you, while allowing TrackerAlignment to depend on Production gives you more flexibility. In the peers scheme, if there is a modified reco module that needs to part of the alignment reco job, it would have to move from TrackerAlignment to Offline, in order for Production to access it.

Muse only understands a linear dependence order, so TrackerAlignment is/will be above Production in dependency, so it will be ready if the peers model runs into this issue.

Or maybe you are saying that the idea that pass1 can only depend on Production and Offline is the primary principle, so this discussion is moot.

brownd1978, 2021-07-20:
Talking with Richie I learned my model for splitting is too simplistic.
Richie needs to iterate running reco to account for 2nd order effects, so
TrackerAlignment should be above Production.  We can remove the loop by
moving one fcl file and maybe 1 filter.   Testing isn’t possible right now
as Minuit is broken in art 3_09.

We probably will want to run the module that produces the millipede data in
pass 1, as that dataset is large.   Do you see a problem with that?  I
think it just means the pass1 musing needs Offline, Production and
TrackerAlignment.  We also need to define a data type for the Milipede
format.
Dave

rlcee, 2021-07-20:
> Talking with Richie I learned my model for splitting is too simplistic. Richie needs to iterate running reco to account for 2nd order effects, so TrackerAlignment should be above Production. We can [...]

I think this is fine

>I think it just means the pass1 musing needs Offline, Production and TrackerAlignment. We also need to define a data type for the Milipede format. 

Also all OK.  I imagine there will be a "Pass1" Musing

kutschke, 2021-07-20:
This all works for me.

rlcee, 2022-09-13:
This was finished up on #426