Mu2e/Offline #627: BinnedSpectrum has physics content, is easily misconfigured

brownd1978, 2021-10-15:
The BinnedSpectrum class currently allows choices for physics options (endpoint, spectra, etc).  These options can be inconsistent with the calling module, and so should be eliminated or moved to the module.

BinnedSpectrum reads a simple data file of the histogram contents without reference or access to their meaning.  This allows misconfiguration, say associating the DIO spectrum with CeEndpoint process.  The spectrum file content needs to be augmented so that the physics assumptions are explicit (what target, what process), and can be validated when read, and applied directly in the module without risk of user misconfiguration.

rlcee, 2022-10-31:
When I brought this Issue up at the meeting on 10/26/22, the answer was that we only need to move the root file to cvmfs.  Rereading this, it seem more is being required.  Please comment @kutschke @brownd1978.  I have the root move prepared.

brownd1978, 2022-10-31:
Here's the comment in the current code:
https://github.com/Mu2e/Offline/blob/882edb85c01f5d987e29f96f4c31a676c45df3ab/Mu2eUtilities/inc/BinnedSpectrum.hh#L163
so there is only a convention, not a code solution.  Using root histograms
solves this with minimal effort for new applications.
Dave

gaponenko, 2022-11-01:
On Mon, 31 Oct 2022, David Brown wrote:

> Here's the comment in the current code: [...]

At the cost of bringing in huge ugly ROOT dependency to the otherwise
simple code.  My preference is to stay away from ROOT whenever
possible, especially in the "core" (as opposed to "analysis") code.
Art might even use a non-ROOT file format in the future, then
the only ROOT dependencies would be through our code.

Generator spectra more often come in the form of text files than ROOT
histograms, so one would need to do a format conversion solving the
same bin boundary problem.  And using ROOT does not address the
spectrum-to-physics association problem raised in the issue, while it
can be easily dealt with using either plain or structured text (XML,
JSON, ...).

Andrei

kutschke, 2022-11-01:
I think it's important that the spectra files be self describing in the sense mentioned above.  I also prefer that the format not be root.  The only question in my mind is if we schedule the work or treat it as a target of opportunity.

rlcee, 2022-11-01:
@kutschke @gaponenko should these be in fcl, json/yaml, XML,  simpleconfig or ? needing description basically eliminates Mu2eUtilties/Table.hh,

kutschke, 2022-11-01:
> @kutschke @gaponenko should these be in fcl, json/yaml, XML, simpleconfig or ? needing description basically eliminates Mu2eUtilties/Table.hh,

I vote against SimpleConfig.  I don't believe that XML supports arrays of numbers without wrapping each number in tags but I may be wrong.  Do we have existing code that reads json or yaml?  If not then fcl looks like the right way to go.

rlcee, 2022-11-01:
XML with a tag per bin is OK with me (its rare and low-volume).  I can't find c++ support for json except some well-known personal github. I could add that as a UPS product or in Offline.  See [https://github.com/open-source-parsers/jsoncpp](url) or [https://github.com/nlohmann/json](url).  I strongly favor one or the other (json or xml) consistently, but right now xml comes from TVMA and genreflex, but json comes from SAM, other tools, and all things python.   Maybe fcl is the best for now.