Mu2e/Offline #662: review number of genreflex targets

rlcee, 2021-11-27:
1) I propose to remove these
GeomPrimitives/src/classes.h
GeometryService/src/classes.h
Mu2eHallGeom/src/classes.h
ProductionSolenoidGeom/src/classes.h
ProductionTargetGeom/src/classes.h
ProtonBeamDumpGeom/src/classes.h
Mu2eInterfaces/src/classes.h
ExtinctionMonitorFNAL/Geometry/src/classes.h

2) looks like it isn't doing anything - propose removing
CalPatRec/src/classes.h

3) experts should comment - split, deprecate, keep?
TrackerMC/src/classes.h - IonCluster
TrkDiag/src/classes.h - ntuples
TrkReco/src/classes.h - KalRep

4) maybe split to a separate repo?
Sandbox/src/classes.h

5) required for interactive root use of the classes, is this a desired feature?
Validation/root/classes.h

6) I'm not sure why dictionaries are needed in this application (for menus?)
EventDisplay/src/classes.h
TEveEventDisplay/src/classes.h

7) standard locations
DataProducts/src/classes.h
MCDataProducts/src/classes.h
RecoDataProducts/src/classes.h

kutschke, 2021-12-02:
Please go ahead with 2. 

About 1.  At one point @gaponenko  was using this to do geometry scans for ExtMon studies ( each grid job generated a different geometry and he recorded the geometry as a subrun (run?) product and used that in downstream code.  Please check with him. Once that is resolved, please send an email to mu2e-hn-sim saying that, if there are on objections. you will remove them some definite date, maybe Monday next week at noon.

gaponenko, 2021-12-05:
Hi Ray,

> Please go ahead with 2. 
> 
> About 1.  At one point @gaponenko  was using this to do geometry scans for ExtMon studies ( each grid job generated a different geometry and he recorded the geometry as a subrun (run?) product and used that in downstream code.  

Yes.  I still think it was really convenient for geometry scans and
eliminated the user error in matching the correct geometry settings
for downstream stages.  However we never wanted to use this in
"production" and the code was not maintained.  I am fine with removing
the dictionaries.  Reducing the build time, clutter and disk use are
nice.

Thank you for doing this,
Andrei

rlcee, 2022-08-25:
1 and 2 completed in #671

rlcee, 2022-10-10:
I'm still looking for input on items 2 through 6.  @brownd1978 please comment on 3, @kutschke please comment on 4, 5 I'll leave as-is for now if no one cares, @ehrlich-uva please comment on 6 (or should we split this directory to its own repo?)

ehrlich-uva, 2022-10-10:
Hi Ray,

I can only comment on EventDisplay (not TEveEventDisplay). 

This is a long time ago when I added this dictionary, but as far as I remember, it became necessary when making shapes (polylines, boxes, ...) used to display tracks, CRV bars, straws, ... right-clickable and change the context menu to add menu items and associated actions.

Ralf

brownd1978, 2022-10-10:
experts should comment - split, deprecate, keep?
TrackerMC/src/classes.h - IonCluster
  used for detailed simulation diagnostic tuple: keep
TrkDiag/src/classes.h - ntuples
  used for diagnostic tuples: keep  
TrkReco/src/classes.h - KalRep
  used for diagnostics, but kalrep is deprecated so these can be eventually removed.

I don't see a big benefit in purging these diagnostics from the schema, or much cost in keeping them, but maybe I'm missing something?

rlcee, 2022-10-10:
> but maybe I'm missing something?

The idea was just to simplify and streamline the build - as you can see, I was able to remove 1 and 2.  It is fine to keep yours as long as useful.  Thanks

rlcee, 2022-10-10:
> I can only comment on EventDisplay (not TEveEventDisplay).

@ehrlich-uva  how do you feel about splitting this subdirectory off to its own repo?  The main advantage is that we are not building it when it is not used.  (I can do the split and preserve history..)  This was the main driver of going to the Muse structure.

kutschke, 2022-10-24:
> I'm still looking for input on items 2 through 6. @brownd1978 please comment on 3, @kutschke please comment on 4, 5 I'll leave as-is for now if no one cares, @ehrlich-uva please comment on 6 (or should we split this directory to its own repo?)

 I am inclined to move Sandbox into it's own repo.  The question is in which organization: I am leaning to putting it in my own GitHub organization.  Does anyone feel it should remain in Mu2e?

rlcee, 2022-10-24:
I suspect there is plenty of value in Sandbox.  For example, you have random tests in there, could they be used to test the random number save and restore, or new random styles (in another issue).  I'd like to continue to have access, but if that's from your account, that's Ok with me.

kutschke, 2022-10-24:
I had forgotten about the Random tests.  It can stay in Mu2e.  If I don't hear back by tomorrow PM that's where it will go.

kutschke, 2022-10-25:
I have created the Sandbox repo, put PR #873 into Offline and added issue 73 to Muse (https://github.com/Mu2e/Muse/issues/73).

closed 2022-10-26 by rlcee