Mu2e/Offline #741: Aggregate behaviour for SubRun and Run Products

kutschke, 2022-03-23:
We need to visit all run and subrun data products that are inserted at end(Sub)Run time.  With the port to art v3_11 we have event range qualified the subrun products as art::fullSubRun(); we do not currently put any products into runs at endRun time. This is good enough for our existing workflows.

Kyle points out that if we want to use MT to its best advantage we will need to support art::subRunFragment().  See below for details.

 Mackenzie has compiled the following list of modules that put products into the SubRun at endSubRun time.

Offline/EventGenerator/src/CORSIKAEventGenerator_module.cc
Offline/EventGenerator/src/CRYEventGenerator_module.cc
Offline/EventGenerator/src/GenEventCounter_module.cc
Offline/EventMixing/src/Mu2eProductMixer.cc (2 places here)
Offline/Filters/src/CompressPhysicalVolumes_module.cc

Kyles advice, taken from an email thread is:

> And yes, it's good to have atomic subruns.  However, you should let the nature of the subrun product determine whether you call art::subRunFragment() or art::fullSubRun().  Once we allow the creation of subrun products from replicated producers, the framework will rely on the ability to aggregate schedule-specific products to form a product that represents all of the subrun's processed events.  The subrun is still stored atomically in the file, but the product will be determined by aggregating the schedule-specific products.  If you specify art::fullSubRun() for such products, you will not get a product that represents the full subrun, but rather the product that represents the events processed by that one schedule.
>   
> Bottomline, my recommendation is to let the nature of the product determine whether it should represent a fragment or a full subrun.  If my explanation above was confusing, I'm happy to clarify.
> 
>

kutschke, 2022-03-23:
The documentation  for aggregation is at: 

  https://cdcvs.fnal.gov/redmine/projects/art/wiki/Product_aggregation_details
  https://cdcvs.fnal.gov/redmine/projects/art/wiki/Example_of_product_aggregation

rlcee, 2022-10-10:
The work for the main point here was done on #742  

For the future possible MT mode for modules producing SubRun products, I'd suggest crossing that bridge when we come to it since enabling MT will be a fairly large project and this can be one of the multiple steps.  @kutschke if you want aggregate method for these products now, I can do that too.

kutschke, 2022-10-24:
>  @kutschke if you want aggregate method for these products now, I can do that too.

No need to do it now.  Let's do it when we know what we really want.  It's important to keep this issue in mind the next time we design/revisit run and subrun products a simulation campaign.