Mu2e/Offline #745: Repeatability of PrimaryProtonGun_module.cc

kutschke, 2022-03-26:
EventGenerator/src/PrimaryProtonGun_module.cc is an art::ReplicatedProducer

When run in MT mode, this does not produce repeatable results because of how environmental effects interact with the scheduler; that is, Event N in one job might land on instance 1 of the module and in a different job it might land on instance 2.  These will have different random number state and produce different events.

For physics purposes this is OK.  But we need repeatability for debugging purposes.

rlcee, 2022-03-26:
Wasn't this module made a ReplicatedProducer for an early version of the geant MT mode, and now it would fine to downgrade to legacy module, and only run one copy?  Isn't this sort of state problem the main reason for not making a module MT?

kutschke, 2022-03-26:
Reverting to legacy feels like a step backwards but I think you are right that it will solve both this issue and and #744.  I will test; I don't think that it will have serious execution time consequences but I will measure to be sure.  If it's good enough I will revert it.

rlcee, 2022-08-26:
@kutschke I propose closing this and #744 and let this module be addressed when/if we review modules generally for promoting to thread safe.  This one only sticks out since it was used in a the early, now obsolete, form of G4 threading.

rlcee, 2022-09-13:
Since there is no need for this module to run multi-threaded, we will leave it as a legacy module and this issue is solved.

closed 2022-09-13 by rlcee