Mu2e/Offline #766: Remove obsolete features of BField support

kutschke, 2022-04-15:
Many of the features below are obsolete for Mu2e but some may be of interest for Mu2e-II.  The job is to remove everything that is truly obsolete and clean up any features that we want to retain for Mu2e-II.

1. We still support reading MECO GMC format maps.  We no longer need that
2. The PARAM format is still used for the Northwestern format of expansion in Spherical Bessel functions
3.  Is the analytic gradient option still useful?
4. Do we still want to support the MECO interpolation style.  It had a problem with field discontinuities when the field point crossed grid boundaries.  The "trilinear" interpolation style is continuous but it has discontinuous derivatives when the field point crosses grid boundaries.  I think that we should remove support for it but anticipate that we may have future, improved interpolation algorithms.
5. Is the option for a perfect solenoidal field in the Tracker region plus real field elsewhere still useful?   I think so - it can be useful if we want to ask if tails in tracking come from imperfect treatment of non-uniform fields or from other sources.
7. There is a single global scale factor.  Do we need to subdivide this, for example, into DS/TS/PS?  For data we may need a Prodition that follows magnet currents?  
8. Should we move the "writeBinaries" option to a separate executable?  My first thought is "no".

rlcee, 2022-09-27:
In discussion with Rob:

1. remove
2. keep
3. keep
4. remove
5. keep
6. skip
7. skip

brownd1978, 2022-09-27:
I don't know if this deserves its own issue, but I am stuck currently in
KinKal with the fact that BFields have limited spatial validity (ie the
field has no value for some input positions), but there is no way to inquire
what the valid range is.  in Mu2eKinKal I need to define what range of fit
is possible, given the configuration.  I call getBFieldWithStatus inside
the fit to see if a point is/is not valid,
but if it is invalid, the only option is to abort the fit.  I currently
hard-code the radial and Z range allowed by the DS field map so I can avoid
extending the fit into the unmapped regions.

kutschke, 2022-09-27:
The information that's potentially available is to return ( xmin, xmax, ymin, ymax, zmin, zmax) for a named map, all in Mu2e coordindates.  We can transform this into other representations if that's useful, for example rmax and zmax (with implied zmin=-zmax) in tracker coordinates, where rmax is that of the inscribed right circular cylinder within the rectangular prism.

If only a single innermap is loaded then calling the function without a map could be implemented.  That function would throw an exception if more than 1 innermap is loaded.

Is that information useful to you?

brownd1978, 2022-09-27:
Zmin and zmax would definitely be useful: i need to extrapolate to the
calorimeter or (eventually) back to the target.  ymin and ymax would be
useful, as I could interpret them radially.
How would I access these?  BFManager doesn't have a 'named map' interface.
And how would I know which named map is appropriate for (say) z limit near
the calorimeter?

kutschke, 2022-09-27:
> How would I access these? BFManager doesn't have a 'named map' interface. And how would I know which named map is appropriate for (say) z limit near the calorimeter?

That's the design discussion that we need to have.  BFieldManager does have:
 https://github.com/Mu2e/Offline/blob/6800e132f2ac1f0657b85d6dc60e0b7c8bac607c/BFieldGeom/inc/BFieldManager.hh#L76

So we could add a function, maybe a free function, maybe a member function of BFManager that:

1. Is named something like getDSMapLimitsInTrackerCoordinates()
2. Accesses this list, checks to see that there is exactly one map with DS in it's name, or some other heuristic.
3. Returns the limits to that map in whatever representation is convenient.

rlcee, 2022-09-27:
It looks like it should be possible to provide a point and return "largest radius you can go from this point", so you could call it for detector 0 to get DS radius and calo z to get that radius.   If outer maps are active, like world map, would you want that included or just inner maps?  That could be a bool input too.

kutschke, 2022-09-27:
The issue with my proposal was always how to specify a map.  Ray's idea will work very well for Dave's use case.  It does require him to look up detector 0 in Mu2e coordinates; Dave: do you naturally have that information?  If so this would be a good way to go.  If not then you may have other ideas.

brownd1978, 2022-09-27:
if possible to implement, "largest distance you can go from point 'x' in
direction 'd'" would be the most general.  Id really like to not have to
enter a named field map: can BFManager internally loop over all the inner
maps?
in reco we only care about the DS, so maybe this is moot.

kutschke, 2022-09-27:
I think that a truly general solution would be very difficult to do well and safely.  I don't see an interesting use case anywhere between yours and the fully general case.  If no one else can see one let's solve your problem explicitly and make it clear that it's not a general solution.

kutschke, 2022-09-30:
Here is a concrete proposal.

I will create a class that in the c'tor has uses GeomHandle to get the BFieldManager.  From there it will get the list of innermaps and determine which map contains the origin of the DetectorSystem.  If there is no such map it will throw an exception.  It will save xmin,xmax,ymin,ymax,zmin,zmax all in Mu2e coordinates.    If you know what accessors you want, let me know.  If not, you can add them as needed.  For example this class could do all of the gymnastics related to coordinate system transformations and box to cylinder.  That way the KK code just asks for the representation it needs.

I suggest that you call this in beginRun.  If you hold it as a data member, update it every beginRun.  Once the BField moves into Proditions we can revisit this, maybe make a Prodition for it?

rlcee, 2022-10-03:
I'd like to get Dave's accessor designed before Rob leaves, so I can implement it while he's gone, if Rob doesn't get to it tomorrow.  Then I can close this issue.  I can make the class Rob is describing, but I think there is a simpler solution of a method
`BFieldManager::getMapRadius(double z, bool onlyInnerMaps=true)`

brownd1978, 2022-10-03:
Hi Ray and Rob,
I really just need the upper and lower X, Y, and Z range of the DS (inner
maps).  I'm not sure what MapRadius would return: 3D distance, Z, X, ???

I don't care what coordinate system it's in, I can convert it to detector.
Rob, If I understood right, the reason you want to defer access to beginRun
is to make sure it isn't affected by module construction order randomness,
is that right?j

Dave

kutschke, 2022-10-03:
Hi Dave,

    The maps come from the geometry service which is not loaded until beginRun time ( technically preBeginRun, meaning before any module beginRun is called).
This was chosen in the very early days when we thought that the GeometryService would have run dependence.    We could decide to change that now that we have decided that the GeometryService holds only the nominal geometry, not the as-calibrated geometry.

     I will make a PR shortly

    Rob

kutschke, 2022-10-04:
Hi Dave,

  The bare minimum is in.  Feel free to extend the class to add accessors to transform to the detector system.  That will keep the client code cleaner.

Rob

kutschke, 2022-10-04:
The PR is in.

rlcee, 2022-10-05:
This last issue, the accessor for the map size, for KinKal, was completed in #863, so this Issue is complete

closed 2022-10-05 by rlcee