Mu2e/Offline #782: CaloClusterMaker DeltaT

gaponenko, 2022-04-27:
Per page 5 of docdb-41910 and discussion in the meeting the physics output for a given event is affected by how events are split into files or subruns.   Since DeltaT is reset at file boundaries events seen early in the file are reconstructed differently from events later in the file.  This is wrong and should be fixed.

rlcee, 2022-08-25:
@gianipez @NamithaChitrazee @bechenard did this time problem ever get fixed?  Thanks

NamithaChitrazee, 2022-08-29:
@rlcee Sorry for the delay in the reply. The issue was fixed temporarily with PR #760. 
But I do not know if further work was done on this to resolve the DeltaT is reset at file boundaries issue. @bechenard would know better.

rlcee, 2022-09-01:
Bertrand asked that this stay open at this time.

brownd1978, 2022-10-11:
The term 'file' in this discussion was a proxy for 'subrun'.  We agreed
long ago not to sub-divide conditions below the sub-run level (or, if finer
dependence is needed, it would be embedded in the conditions object fetched
at begin_subrun).
So I don't think there's a conceptual problem.  There is an implementation
issue that the calorimeter code should be fetching delta-T (and related
values) from the database, not using hard-coded values.  That has
already been discussed and agreed to by the authors.

bechenard, 2026-02-12:
As written by Dave, the deltaT value is fetched at begin subrun, which is also a boundary of between conditions. The implementation in the database will be addressed in another issue.