Mu2e/Offline #845: Code that might fail if our staging system evolves

kutschke, 2022-09-02:
The following lines of code contain tests that work because they are an accident of our current system of managing staging.  They are fragile and could break with future evolutions.  These should be made more robust.

https://github.com/Mu2e/Offline/blob/72ad4f72f5fb125135417b9d2d2b896af1a96823/Analyses/src/PbarAnalysis2_module.cc#L531

https://github.com/Mu2e/Offline/blob/72ad4f72f5fb125135417b9d2d2b896af1a96823/Analyses/src/PbarAnalysis2_module.cc#L542

gaponenko, 2022-09-02:
The code in question is already broken.   Since March 2021, commit #9294cc21fb4eccd73840026edf8644550a109e15

rlcee, 2022-09-02:
I suspect a notable amount of code, especially in Analyses, has degraded, become obsolete, or been superseded.  Do we have any rules?  It must compile, but must it be maintained correct and functional? I suggest letting Bob know and closing this Issue.  Maybe a better Issue is to plan for splitting off single-user code to personal repos and slimming Analyses significantly.