Mu2e/Offline #849: Robust random seed plan for digitizers

rlcee, 2022-09-13:
Currently only geant  (Mu2eG4) module runs multi-threaded.  In order to keep random numbers reproducible, each event has its random number reseeded by the event number (plus salt?).  So whichever thread receives an event, the simulation will still reproduce.  If the events are then written out, and the next stage reads the events back in, art causes the events to enter the new path sorted by event number, and therefore if random numbers are used by single-threaded modules in this second stage, the result is still reproducible.  I think this is what is happening in our current procedures.

In the case that there are modules which use random numbers and are in the same job as a multi-threaded geant and downstream of geant in the path, then those modules will get events in a semi-random order, their random chains will be unpredictable, and the result of these modules will be not be reproducible.  A notable case is if the digitizers, which use random numbers, would be active in the same job as a multi-threaded Mu2eG4.  Further note that ceSimReco does run digitizers in the same job as geant, but it uses single-threaded geant, so will not have this problem.

A potential solution is to run the digitizers with random numbers seeded by the event number, like the geant module.  This might be able to be implemented as a feature of the random number service.

pavel1murat, 2022-09-29:
an issue with the RN stream initialization exists even in a single-threaded environment. Modules using RN's and getting their initialized RN streams from the SeedService  "interfere" with each other,  so the RN's used by a module, in an obscure way, depend on the presence of other users of the SeedService 

In a single-threaded mode, this issue could be addressed by making the starting RN stream for a given module a function of the module label, instead of the order in which art initializes the modules

rlcee, 2022-10-18:
Initalizing CLHEP HepJamesRandom (setSeed method) 1M times implies one intialization takes about 26e-6s. It involves several times 10k arithmetic operations.  A ceSimReco event takes  0.2s, so intializing 20 random engines every event would add 0.25% to the CPU time.

brownd1978, 2022-10-18:
ceSimReco isn't a production job, more relevant is pure signal
digitization, which is ~0.1 second/event.  Mixing takes a few
seconds/event.  So again the same conclusion.
As long as we can avoid unnecessarily creating seeds for modules which
don't need them this doesn't seem a problem.

kutschke, 2022-10-25:
Hi Ray,

  Did you do this with a single 32 bit seed?   When you get a chance can you remeasure using seeds of length 2, 3, 4 longs?

Rob

rlcee, 2022-10-25:
in units of microseconds per call
HepJamesRandom setSeed(long) = 26
MixMaxRng setSeed(long) = ~0
MixMaxRng setSeeds(long*,1) = 26
MixMaxRng setSeeds(long*,2) =  51
MixMaxRng setSeeds(long*,3) = 75
MixMaxRng setSeeds(long*,4) = 101

gaponenko, 2022-10-25:
The scaling of MinMaxRng seed setting is surprisingly efficient.  (For 
those who remember RANLUX.)

Andrei