Mu2e/Offline #851: Random Number state restore

kutschke, 2022-09-14:
The RandomNumberGenerator service has capabilities to restore engine state at the start of each event.  The existing documentation is at:

 https://mu2ewiki.fnal.gov/wiki/RandomNumbers#Saving_and_Restoring_State

This has not been tested for a long time and I don't know if it has ever been tested following the transition from HepJamesRandom to MixMax.

The job is to

1.  Understand what works and what does not
2.  Work with the art team to restore any missing functionality that we need
3. Update documentation and provide standard examples.

brownd1978, 2022-09-14:
There is a larger issue, that we need a way to reproducibly seed all the
engines in an event based just on the EventId, or some equivalent payload.
Do you think that belongs in this issue?

kutschke, 2022-09-14:
> There is a larger issue, that we need a way to reproducibly seed all the engines in an event based just on the EventId, or some equivalent payload. Do you think that belongs in this issue?
> […](#)

I think Dave's suggestion fits better in #849 .  @rlcee what do you think?

brownd1978, 2022-09-15:
Yes, 849 covers my comment.  but in that case, do we need 851?  In other
words, when 849 is addressed, won't the technology behind 851 be obsolete?

kutschke, 2022-09-15:
I think that 849 and 851 cover different aspects: 849 covers seeding at the start of the job while 851 covers saving and restoring the state (not the seed) mid job.

rlcee, 2023-08-17:
The random number restore has be validated with the following steps
- run 100 events of potSim, writing out the random number record and all products
- filter out second 50 events, all products, make validation histograms on this file, this is result A
- filter out last 50 events writing only the random number record, call this file B
- run potSim with rootInput module running on file B and restore random turned on, run validation of the output file, this is result C
- result A and C are identical
This was pretty straightforward.  I used potSim because it is using only EmptyEvent on input, while ceSimReco is using a stops auxiliary input file, which would complicate the test.  The only trick was that in restore, you can't skip `defaultEngineKind: "MixMaxRng"` which is implied in the current wiki instructions.  I will update that, then I think we can close this Issue.

closed 2023-08-25 by rlcee