Mu2e/Offline #852: SimParticle

brownd1978, 2022-09-15:


brownd1978, 2022-09-15:
@FNALbuild run build test with Mu2e/Offline#850

rlcee, 2023-07-17:
@brownd1978 this looks tedious but straightforward.  Please confirm this issue has not changed.

brownd1978, 2023-07-17:
Yes this is still valid.  There's one use case where SimParticleTimeMap is
still used, namely in CosmicTimeOffset, but that's as an event product.

rlcee, 2023-07-19:
> > Yes this is still valid. There's one use case where SimParticleTimeMap is still used, namely in CosmicTimeOffset, but that's as an event product.
> 
> CosmicTimeOffset (and cosmics generators) seems to produce a SimTimeOffset object, which seems to be different from SimParticleTimeMap. Is SimTimeOffset an unrelated object?

@brownd1978 is there an easy answer to this question?  Thanks

brownd1978, 2023-07-20:
Yes you're right, I was confusing SimTimeOffset with
SimParticleTimeOffset.  The latter is the one that depends on
SimParticleTimeMap.  It needs to be removed along with.

Dave

rlcee, 2023-07-20:
OK, if I understand you correctly, I will continue to remove SimParticleTimeMap product and SimParticleTimeOffset utility class.

rlcee, 2023-09-05:
The following fcl mention timemaps or "protonTime". Note that all these are in test dirs or in old CRV fcl. Ideally, the expert who knows the purpose of this fcl would update the fcl to a coherent modern state.  Since that is unlikely to happen, we have some less preferred choices.  We could assign each to an experts to fix within a given deadline or we will delete it.  We can ~blindly chop out these references, leaving the fcl in a unknown state, or leave these references in so if they are used again, the user will see "no such module" type of error to give a clear clue on what to do to update the fcl.  I propose the latter as most logical and least disruptive.   With no further comment, I will close this issue on 9/7.

CRVResponse/fcl/prolog_v03.fcl
CRVResponse/fcl/prolog_v05.fcl
CRVResponse/fcl/prolog_v06.fcl
CRVResponse/fcl/prolog_v07.fcl
CRVResponse/fcl/prolog_v08.fcl
CRVResponse/fcl/prolog_v10.fcl
CRVResponse/fcl/prolog_v09.fcl

Analyses/test/debug_config_bug.fcl
Analyses/test/genMixReco.fcl
Analyses/test/mixPointsDump.fcl
Analyses/test/readCaloDigi.fcl
Analyses/test/readTrackCluster.fcl
BFieldGeom/test/makeBinaryMaps.fcl
BFieldGeom/test/readBinaryMaps.fcl
CaloMC/test/RunCaloCalibGun.fcl
CalPatRec/test/calPatRec.fcl
CalPatRec/test/calPatRec_01.fcl
CalPatRec/test/calPatRec_02.fcl
CalPatRec/test/calPatRec_display.fcl
CalPatRec/test/calPatRec_display_cnv00502.fcl
CalPatRec/test/calPatRec_display_read.fcl
CalPatRec/test/mergePatRec_02.fcl
CalPatRec/test/mergePatRec_03.fcl
CalPatRec/test/mergePatRec_display.fcl
CalPatRec/test/mergePatRec_display_read.fcl
CalPatRec/test/trkPatRec_01.fcl
CalPatRec/test/trkPatRec_02.fcl
CalPatRec/test/trkPatRec_display.fcl
CalPatRec/test/trkPatRec_display_read.fcl
CRVResponse/test/CRVResponsePlot_v05.fcl
CRVResponse/test/CRVResponse_RadiationBackgroundStudies.fcl
CRVResponse/test/CRVResponse_v05.fcl
CRVResponse/test/CRVResponse_v07.fcl
CRVResponse/test/singleModule/singleModule.fcl
CRVResponse/test/singleCounter/singleCounter.fcl
CRVResponse/test/singleCounter/singleCounter_darkNoise.fcl
CRVResponse/test/singleCounter/singleCounter_point.fcl

CaloMC/test/RunCaloCalibGun.fcl:      @table::CommonMC.producers
CRVResponse/test/CRVResponse.fcl:    @table::CommonMC.producers
CRVResponse/test/CRVResponseDAQ.fcl:    @table::CommonMC.producers
CRVResponse/test/CRVResponseReco.fcl:    @table::CommonMC.producers
Mu2eKinKal/test/TTCPR.fcl:    # @table::CommonMC.producers

closed 2023-09-07 by rlcee