Mu2e/Offline #862: Differences in the production of the conversion electrons

Hiigaran, 2022-09-30:
Good morning,

In Mu2e/Production/Validation/ we found two procedures to generate the conversion electrons.

They give different results as kinematics, trajectories and identities of the simulated particles. We identified the origins of the differences (see slides attached).

[Differences in the production of the conversion electrons.pptx](https://github.com/Mu2e/Offline/files/9684226/Differences.in.the.production.of.the.conversion.electrons.pptx)

kutschke, 2022-09-30:
Thanks.  Please provide the following.

1. Which version of Offline are you using.  Provide the tag or SHA of the head commit. 
2. Two fcl files to run the jobs, one for each configuration
3. If step 2 just makes output files, and no printout, please provide 2 more fcl files to read the output files and make the printout that demonstrates the problem.

Let me know if you have any questions.

Hiigaran, 2022-10-03:
[Examples.tar.gz](https://github.com/Mu2e/Offline/files/9697888/Examples.tar.gz)

rlcee, 2022-10-10:
@kutschke @brownd1978 @gaponenko @pavel1murat What work needs to be done here?
1) the differences in the two CE fcl methods was tracked to the difference in the field.  This is consistent with a particle entering the TSd and altering the random number chain for following events.
2) differences due to adding modules in the path. This is also consistent with random numbers being assigned differently.

Since these are expected effects in the current code, I'd suggest closing this Issue, and improvements can be discussed on #849
(Pasha already commented)  Or do these random number explanations need to be proven?

pavel1murat, 2022-10-11:
I think, in the original consideration, we confused bfgeom_reco_v01.txt with bfgeom_v01.txt and assumed that it includes all maps - at least I did.

Perhaps, it would make sense for Production/Validation/ceSimReco.fcl job to use "Offline/Mu2eG4/geom/bfgeom_v01.txt" ?
Would that affect the validation memory footprint to the extent requiring modification of the workflows?

rlcee, 2022-10-12:
> Perhaps, it would make sense for Production/Validation/ceSimReco.fcl job to use "Offline/Mu2eG4/geom/bfgeom_v01.txt" ? 

I'm fine with this, I'd like to ask @brownd1978 to sign off

>Would that affect the validation memory footprint to the extent requiring modification of the workflows?

ceSimReco already requires over 2GB of memory, so more than one grid slot, so that won't change.

brownd1978, 2022-10-12:
I’m ok using the larger map.  Ray, can you run a test to find the new HWM?
  I’d like the jobs submitted with an accurate memory estimate, as Rob
claims fractional slots can be used by other jobs.

Dave

rlcee, 2022-10-12:
> can you run a test to find the new HWM?

it goes from 2798 to 3069 MB

brownd1978, 2022-10-12:
OK.  Maybe submit with 3200MB?

rlcee, 2022-10-12:
> OK. Maybe submit with 3200MB?

Validation was at 3000 and I moved it to 3300, but I don't know any hard rules on the variation that has to be accommodated.

kutschke, 2022-10-14:
I am OK with the memory change but it's won't solve the problem.  See  comments that I just added at: https://github.com/Mu2e/Production/pull/206

kutschke, 2022-10-14:
@pavel1murat As I understand the discussion above, you now believe that the changes in the field maps and the seeding algorithm adequately explain the differences in results.  Is that correct?  Or do you believe that there are still mysteries to be solved?

kutschke, 2022-10-15:
> Validation was at 3000 and I moved it to 3300, but I don't know any hard rules on the variation that has to be accommodated.

Is this the use case to push us to set up the tooling to plot timelines of resources used by validation: min,  max, mean and rms  over the N jobs for each type of validation?

pavel1murat, 2022-10-16:
> @pavel1murat As I understand the discussion above, you now believe that the changes in the field maps and the seeding algorithm adequately explain the differences in results. Is that correct? Or do you believe that there are still mysteries to be solved?

- that is correct. I didn't appreciate that ceSimReco  job was using a truncated BField map.

rlcee, 2022-10-21:
The field used in ceSimReco was made the same as the field in ceSteps in Mu2e/Production#206  The issues of coordinating random seeds are being addressed in #849  As decided in the comp/soft meeting 10/19/22, we are closing this issue.

closed 2022-10-21 by rlcee