Mu2e/Offline #891: Remove artdaq includes from the list of -I options seen by all files

kutschke, 2022-11-10:
Dave pointed out that, when we upgraded from envset p034 to p035, we only changed versions of artdaq_core_mu2e and pcie_linux_kernel_module.  However this triggered a recompile of every .cc file in Offline even though the only files that use these external libraries are in DAQ/src

The reason is that both ADCM and PCIE are included in the standard include path defined by the function cppPath in python/sconstruct_helper.py .  scons recognizes that, for every file, the include path has changed from one build to the next; therefore it recompiles every file.

The minimal solution has two parts
1. Remove these two ups products from the standard include path
2. Add them as include path extras in DAQ/src/SConscript

There may also be some other SConscript files, not in Offline, that need to be updated.

kutschke, 2022-11-10:
The piece for which I don't immediately see a clean solution is how to update Muse in a way that the new version of Muse will work both with both before and after versions of Offline.

rlcee, 2022-11-18:
I think when the artdaq version changed, the envset must have changed, and Muse pointed to a new build area, so everything has to be rebuilt.  This is expected in the design.  I believe scons works on path+file dependence, not just path.  To check this, I created a local envset, built, then changed the artdaq version in the local envset, resetup and built again.  In this case, it only compiled the DAQ directory code, and relinked a small subset of libraries.

I will close this on Monday unless someone has more comments.

P.S. Generically, a way to tell Muse if it is working in a new-style build or not is to add a flag to the .muse file.

closed 2022-11-21 by rlcee