Mu2e/Offline #898: StrawDrift needs re-implementation

brownd1978, 2022-11-25:
The current implementation of StrawDrift is based on a representation of a 2d array using a 1d vector with indexing functions.  There is at least 1 physics error in the implementation of the instantaneous drift velocity, which doesn't include the phi dependence.  The rest of the code is almost impossible to understand and debug.  It should be rewritten from the bottom up based on true 2d data containers (vector<vector<>>).

kutschke, 2022-11-28:
Is the issue that the index manipulation is exposed and therefore error prone?  Is it safe to assume that the dimensions are known at compile time and will not change with time? I bet the answer is no.

I recommend that we acquire a class, or class template, that properly manages a single std::vector<T>, of length n1*n2, and use that.  My objection to vector<vector<T>> is that it will, in general, have poor cache performance. 

We already have the 3D version of the class template: BFieldGeom/inc/Container3D.hh  .   If we cannot acquire a better option, we can ask the art team to review that template and then produce the 2D version.

brownd1978, 2022-11-28:
> Is the issue that the index manipulation is exposed and therefore error
> prone? Is it safe to assume that the dimensions are known at compile time
> and will not change with time? I bet the answer is no.
the current code functions but is difficult to decipher and hence maintain.

> I recommend that we acquire a class, or class template, that properly
> manages a single std::vector, of length n1*n2, and use that. My objection
> to vector<vector> is that it will, in general, have poor cache performance.
cache behavior depends on access patterns; the code mostly performs
interpolations along the time  dimension, but it also performs limited
interpolation along phi. I think the best implementation would be
vector<Class>, where Class contains a vector<double> (For time) Plus the
most common interpolation code.  Then StrawDrift itself just performs the
simpler phi interpolation.

> We already have the 3D version of the class template:
> BFieldGeom/inc/Container3D.hh . If we cannot acquire a better option, we
> can ask the art team to review that template and then produce the 2D
> version.
Unlike the BField case the dimensions here aren’t Cartesian (time vs
angle), so a generic 2D implementation won’t help.

> <https://github.com/Mu2e/Offline/issues/898#issuecomment-1329294889>, or
> unsubscribe
-- 
David Brown            ***@***.***
Office Phone (510) 486-7261
Lawrence Berkeley National Lab
M/S 50R5008 (50-6026C)  Berkeley, CA 94720

kutschke, 2022-11-28:
> Unlike the BField case the dimensions here aren’t Cartesian (time vs angle), so a generic 2D implementation won’t help.

If a generic 2D won't work, then my suggestion is moot.   If you are doing interpolations in both directions, then it would still be useful to think about the memory layout.