Mu2e/Offline #906: Reduce unnecessary build-time dependencies

kutschke, 2022-11-30:
This is prompted by an email from Dave in which he reported that changing a single .cc file in TrackerConditions/src triggered a rebuild of over 200 files. 

I experimented and found that the rebuild naturally rebuilds
 
build/sl7-prof-e20-p035/Offline/tmp/TrackerConditions/src/StrawDrift.os
build/sl7-prof-e20-p035/Offline/lib/libmu2e_TrackerConditions.so
 
Then it relinks another 202 .so files.   All of those .so files  have TrackerConditions.so in their link list.

Some of those link list entries are clearly spurious.  For example, Validation does not depend on TrackerConditions or some of the other things in its link list.  I will submit a PR to address this one.

Others are more complicated.  For example the all 86 modules in Analyses were rebuilt even though only Offline/Analyses/src/ReadStrawHitReco_module.cc  uses TrackerConditions.  Similarly all  11 libraries made by DAQ were rebuild even though only 1 of them depends on TrackerConditions.

I used Offline/Analyses/src/ReadStrawHitReco_module.cc as a model to investigate further:

I removed TrackerConditions from the link list and added the ProditionsService to the link list.  The compile works but the link fails because it cannot resolve `typeinfo for mu2e::StrawResponse' .  If I add back TrackerConditions, this is resolved.  The ProditionsHandle template uses the typeinfo to find the StrawResponse object inside of its store of objects.

So the question is if we can reengineer the ProditionsService and ProditionsHandle so that caller does not need to link against the TrackerConditions library just to get the typeinfo.    If we can do this, then ReadStrawHitReco can get access to StrawRespose via the ProditionsService.

kutschke, 2022-11-30:
One more comment: we want to do this while keeping --no-undefined when we build libraries.  This might be a deal breaker.

rlcee, 2022-11-30:
>So the question is if we can reengineer the ProditionsService and ProditionsHandle so that caller does not need to link against the TrackerConditions library just to get the typeinfo. 

Since the purpose of the handle is to deliver StrawResponse objects, the module will always need to link to TrackerConditions.

>If we can do this, then ReadStrawHitReco can get access to StrawRespose via the ProditionsService.

This sounds like forcing the StrawReponse object to be linked in the Proditions .so.  I think that would be a notable change.

It seems another way to go is to allow each module to define its own link list, and not depend on the nominal link list created for the directory.   This also ties in to an idea we've discussed before - proving many more automated link lists so so that a module, or SConscript, could request to link to "art" or "RecoProducts" and all the relevant libraries will be included by the system.   Let me know if I'm missing some point.