Mu2e/Offline #927: Mu2e data model: connected collections

pavel1murat, 2023-01-21:
I'll start with a simple analysis example. The Mu2e reconstruction is using TimeClusters which are made out of ComboHits. Any low-level analysis  dealing with TimeClusters needs to know which ComboHits the TimeClusters are made of.  Therefore an analysis module using TimeClusters in addition to a TimeCluster collection tag needs to be passed an additional tag, a tag of a  ComboHit collection. 

Would it make sense to store in a TimeCluster collection a tag of the corresponding ComboHit collection, or a pointer to it in any other form?  - That would make a task of configuring a job quite a bit and would reduce the number of mistakes made and the time spent on that. 

I think there is a whole class of cases where we could benefit from using the same approach.

rlcee, 2023-01-21:
I'll add two quick comments
- recall Andy's system for this problem, doc 15180
- fairly recently, Kyle added an art pointer which points to a whole product (as opposed to an item in a vector in a product).

pavel1murat, 2023-01-22:
- I think, the above solution is much simpler than ProductBags 
- your option two, a pointer to a collection, could be a good solution for an art-based system
- in general, writing out strings and integers is more flexible and has less side effects than storing poiters of special types, I see that as an advantage for a system which users are not professional software developers

brownd1978, 2023-01-23:
Just to remind people of the context: we found very early on that the
overhead associated with (then existing) art provenance tools (Ptrs) was
too high to use those in low-level data objects like hits involved in
time-critical applications like the trigger, hence why just indices are
stored as reference.  While it's true that those are not a complete
reference (the collection name is also needed), they are sufficient within
the standard reco sequence.  For my information, what sequence are you
running in which the ComboHit origin is ambiguous?

If we're going to re-open this very old question we need to be sure the
need is real, and to prioritize it relative to other work.  I suggest you
make the case at the comp-soft meeting.  If we decide this is needed, then
my strong preference is to use official tools, and not invent something
ad-hoc ourselves.  Adding a collection-level ptr to the TimeCluster might
work but AFAIKvthat technology is untested within Mu2e.  There is also the
stored art::ProductID, which is used to keep track of collections inside
ComboHitCollection itself.  That is known to cause some processing
overheads, but is within acceptable limits.
Dave

kutschke, 2023-01-23:
I started to write a detailed reply but it did not add anything to what Dave just said.    I would like to hear the details of the specific problem that Pasha encountered.

pavel1murat, 2023-02-04:
just try to print a time cluster with all its hits ...