Mu2e/Offline #936: CrvRecoPulses random order

rlcee, 2023-02-13:
Running `Production/Validation/nightly/reco_00.fcl` repeatedly and printing the products shows that CrvRecoPulses are stored in a random order.  The product should be reproducible.   Also I see that  the chi2 field is often a nan, but I believe we should avoid storing nan's in  products.

ehrlich-uva, 2023-02-13:
I have no explanation for the random order. There is nothing random in the CRV reco sequence.
The chi2 gets set to NAN, if the fit fails. In this case, an error flag gets set for the reco pulse and the pulse parameters are calculated in a different way (so that the pulse can still be used).

rlcee, 2023-02-25:
random order addressed in #950  In discussions there was some agreement to change chi2=NaN to chi2=-1

kutschke, 2023-02-26:
> random order addressed in #950 In discussions there was some agreement to change chi2=NaN to chi2=-1

If we do change chi2 to -1, do we need to add protection to the logchi2 plot in ValCrvRecoPulse.cc ?

kutschke, 2023-07-17:
NaN to -1 works for me.  Won't there still be an undefined sort order if there is more than one -1 in the group?  I think that stable sort will resolve that.

ehrlich-uva, 2023-07-17:
I can change it to -1. There are CrvRecoPulseFlags. So I could add an additional flag for these types of chi2s.

ehrlich-uva, 2023-07-17:
Yes, there could be multiple pulses with a chi2 of -1 within a single event. However, each reco pulse has its own chi2 and its own CrvRecoPulseFlags. Downstream modules need to decide what to do with pulses that have a chi2 of -1. So far, no module uses this variable.

kutschke, 2023-07-17:
Do you understand the source of the random order - I thought it was from sorting on chi2 and sorting on a NaN produces undefined behaviour - maybe I am confused.

ehrlich-uva, 2023-07-17:
The random order problem was solved in PR 951. An explanation why we saw this random order was found by Ray - see top of PR 950.

rlcee, 2023-07-17:
@ehrlich-uva I'll assume you will complete this shortly.  If not, I'll ping it in a few days.  I'm happy to help.

kutschke, 2023-07-17:
> The random order problem was solved in PR 951. An explanation why we saw this random order was found by Ray - see top of PR 950.

Thanks for the reminder.

Ray: what's the plan to move forward?  Will you make the change or ask Ralf to?  

Ralf: is there any code that tests for NaNs?

rlcee, 2023-07-17:
It sounded like Ralf was willing to.  I'll come back to it if he doesn't get to it.

ehrlich-uva, 2023-07-17:
Yes, I can make the changes. There is no code that tests for chi2 being NaN, because there is no code (yet) that uses these chi2s.

ehrlich-uva, 2023-07-18:
I just created PR #1014

rlcee, 2023-07-19:
All done!

closed 2023-07-19 by rlcee