Mu2e/Offline #975: Replacing DataProducts/inc/EventWindowMarker.hh

kutschke, 2023-05-01:
The class https://github.com/Mu2e/Offline/blob/main/DataProducts/inc/EventWindowMarker.hh was created to support two pieces of information needed for the MDC2020 simulations.  

1. An on-spill /off-spill flag
2. The event duration, in ns, which will be either 1675 or 1700 ns for on-spill events

The class is ill-named because it does not really have anything to do with the Event Window Marker, which is a timing signal distributed by the CFO.    The on/off spill marker is also a duplicate of information that will ultimately be found elsewhere.

The authoritative source of the on/off spill information is the CFO heartbeat packet and there is a proposal to store a copy of this packet as part of an event header that the TDAQ will add to every art event.  A prototype event header class exists at : https://github.com/Mu2e/artdaq_core_mu2e/blob/develop/artdaq-core-mu2e/Data/EventHeader.hh .  

At present this class only contains the information from the CFO heartbeat packet but it could be extend with other information if that is needed.    For example the event duration is not present.

I believe that the event duration in ticks of the 40MHz DAQ clock will need to be distributed in the CFO heartbeat packet - how else will the readout electronics know what to do?  I will ask Ryan and Eric to comment on whether or not this is correct.  If it proves to be correct, all of the information needed to replace DataProducts/inc/EventWindowMarker.hh  is available in this class.

@rrivera747 @eflumerf   Please comment on the above para.

@bonventre I think you are a stakeholder in this discussion.

bonventre, 2023-05-01:
Hi Rob - one clarification, the electronics does not do anything with the DAQ event duration information, all it gets is the onspill/offspill flag. Then digitization window start/stop are always defined relative to the arrival of the heartbeat, and so are actual event length agnostic. In Offline, EventWindowMarker provides info for MC sim only, and so needs the true event length. ProtonBunchTime is the dataproduct that gives the reco information about the event length for correcting the times, and that could eventually be generated from some processing of the heartbeat packet information.

kutschke, 2023-05-01:
Thanks for the clarification.  I believe that one time that we need to know the event duration is during analysis  when we are deciding on the end of the fiducial time window and the integrated livetime.  Unless someone can show me that we don't need it then I think we should plan to have it.

kutschke, 2023-05-01:
@bonventre Another followup.   As I read page 7 of https://mu2e-docdb.fnal.gov/cgi-bin/sso/RetrieveFile?docid=4914&filename=Mu2e_ROC_Packet_Protocol.pdf&version=37  the heartbeat packet arrives at each ROC sometime prior to the start of the event window.  At a later time the ROCs receive the event window marker and the next rising edge of the DAQ clock defines the start of the event Window.  Is that also your understanding?  And TDC values recorded in the event are relative to the start of the event window - right?

bonventre, 2023-05-01:
Right, I wasn't quite right above - there is the heartbeat packet, which configures the next event, and then the actual marker which arrives later, and the TDC times are relative to the marker as you said.

Then the TDAQ also may provide two different things - first, the length of the event in clock ticks, and second, a measure of the actual time of the event marker relative to the protons (and these both may be provided for the previous event instead of the current one). Nothing that I know of actually uses the first number (although it could be used as a cross check), but we do need the second number for reco, and that is what is in ProtonBunchTime. StrawHitReco for example uses ProtonBunchTime to calibrate the TDC to a time relative to protons.

kutschke, 2023-05-01:
Thanks Richie.  We are on the same page about your first para.

I am in a separate conversation with the TDAQ people about the first part of the second paragraph.  Once that converges I will summarize here.

Does the trigger use ProtonBunchTime or just reco?

bonventre, 2023-05-01:
Right now it looks like hit reco for the trigger (TTmakeSH etc) also uses the ProtonBunchTime from EventWindowMarkerProducer, which currently assumes MC truth smeared by 0.5 ns

kutschke, 2023-05-02:
> Right now it looks like hit reco for the trigger (TTmakeSH etc) also uses the ProtonBunchTime from EventWindowMarkerProducer, which currently assumes MC truth smeared by 0.5 ns

Thanks.  

I learned some new information from Ryan and Greg.   First, the Delivery Ring RF0 TDC that is found in the heartbeat packet is the value predicted for the current event, based on previous events.   I believe that it counts 1.25 ns ticks.  So your 0.5 ns is reasonable ( 1.25/sqrt(12) = 0.36 ).   I can imagine the prediction being off by 0 and +/-1 ticks; I am not sure if +/- 2 is possible but it should be rare.

So this will be the ultimate replacement for ProtonBunchTIme in the trigger.   As you said, nothing in the trigger needs to know about 67 vs 68 tick events.

Second, there is a proposal, not yet implemented, to create a new record called the "CFO Event Window Data Record"

For the content, see item 3 on page 22 of  https://mu2e-docdb.fnal.gov/cgi-bin/sso/ShowDocument?docid=19095 .  This information would be computed by the CFO for each event and distributed to the DTCs.  Note that the record for event N will not be complete until after the arrival of the  EventWindowMarker for event N+1.  The proposal is for the second stage event builder to put this record into the art event ( ie by the same code pulls the CRV info into the Trk+Cal event).

I have asked Ryan about the reason for the repeated information in the payload.

The bottom line is that the trigger needs the RF0 TDC and can use the predicted value from the heartbeat packet.  Using the CFO Event Window record, reco will have the choice of using the predicted or measured value; there are choices to discuss at the right time in the right forum, but not now.  Also, analysis, and possibly reco, will need to know the event duration, if only for defining the fiducial window and normalization, which they can get from the CFO Event Window Data Record. 

I thinks this wraps up the physics content of this issue.  What remains is to decide on names and content for the class(es) that represent the heartbeat packet and the CFO Event Window Data Record; and then to choose a time for the disruptive changes.

What do others think?