Mu2e/Offline #1029: taking clang-tidy seriously

rlcee, 2023-08-03:
Noting that in CI, clang-tidy typically produces dozens of warnings, which makes it difficult to take seriously, this Issue is a plan for reducing the errors and warnings to those which should be examined.  The full clang-tidy log file is /mu2e/app/users./rlc/tidy.log.  A summary is below. 

  4982 [readability-braces-around-statements]
   4120 [bugprone-narrowing-conversions]
    615 [clang-diagnostic-error]
    324 [bugprone-easily-swappable-parameters]
    258 [bugprone-reserved-identifier]
    218 [performance-unnecessary-value-param]
    177 [clang-diagnostic-unused-private-field]
     80 [performance-for-range-copy]
     77 [clang-analyzer-deadcode.DeadStores]
     52 [performance-unnecessary-copy-initialization]
     34 [clang-diagnostic-inconsistent-missing-override]
     26 [bugprone-branch-clone]
     22 [performance-faster-string-find]
     19 [bugprone-incorrect-roundings]
     16 [readability-non-const-parameter]
     16 [clang-analyzer-cplusplus.NewDeleteLeaks]
     15 [clang-analyzer-optin.cplusplus.VirtualCall]
     14 [bugprone-integer-division]
     14 [bugprone-implicit-widening-of-multiplication-result]
     13 [clang-diagnostic-pessimizing-move]
     11 [clang-diagnostic-unused-variable]
     10 [readability-misleading-indentation]
     10 [performance-inefficient-string-concatenation]
     10 [clang-diagnostic-unused-but-set-variable]
     10 [bugprone-macro-parentheses]
      8 [clang-diagnostic-format-security]
      8 [bugprone-parent-virtual-call]
      7 [clang-diagnostic-range-loop-construct]
      7 [bugprone-unused-return-value]
      6 [performance-no-automatic-move]
      5 [performance-move-const-arg]
      5 [performance-inefficient-vector-operation]
      5 [clang-analyzer-security.FloatLoopCounter]
      4 [clang-diagnostic-unused-lambda-capture]
      4 [clang-diagnostic-uninitialized]
      4 [clang-diagnostic-missing-braces]
      4 [clang-diagnostic-dangling-gsl]
      4 [bugprone-too-small-loop-variable]
      4 [bugprone-exception-escape]
      3 [clang-diagnostic-sometimes-uninitialized]
      3 [clang-analyzer-core.CallAndMessage]
      2 [clang-diagnostic-warning]
      2 [clang-diagnostic-self-assign]
      2 [clang-diagnostic-delete-non-abstract-non-virtual-dtor]
      2 [clang-diagnostic-absolute-value]
      2 [bugprone-unused-raii]
      2 [bugprone-fold-init-type]
      1 [clang-diagnostic-vexing-parse]
      1 [clang-diagnostic-unused-value]
      1 [clang-diagnostic-unused-function]
      1 [clang-diagnostic-unused-const-variable]
      1 [clang-diagnostic-tautological-pointer-compare]
      1 [clang-diagnostic-parentheses-equality]
      1 [clang-diagnostic-non-c-typedef-for-linkage]
      1 [clang-diagnostic-mismatched-tags]
      1 [clang-analyzer-optin.cplusplus.UninitializedObject]
      1 [bugprone-throw-keyword-missing]
      1 [bugprone-redundant-branch-condition]

rlcee, 2023-08-03:
>4982 [readability-braces-around-statements]
`    if (X->Panel % 2 == 0) X->Face = 0;`

propose turn this check off - too many examples, hopeless

>4120 [bugprone-narrowing-conversions]
```
narrowing conversion from 'double' to 'float' [bugprone-narrowing-conversions]
           addPulse(pulseMap, crystalID, time, eDep);
```

propose to turn this off as too many examples - hopeless

>615 [clang-diagnostic-error]

This one is all due to BitMap 
```
no member named 'ptr_fun' in namespace 'std' [clang-diagnostic-error]
        cname.erase(cname.begin(), std::find_if(cname.begin(), cname.end(), std::not1(std::ptr_fun<int, int>(std::isspace))));
```

Propose to re-write code

>324 [bugprone-easily-swappable-parameters]
`       HitInfo(float time, float eDep)
`

Propose to turn this off

>258 [bugprone-reserved-identifier]
`    TTree*                     _Ntup;`

Propose to fix these and start a policy against this style

>218 [performance-unnecessary-value-param]
```
parameter 'name' is passed by value and only copied once; consider moving it to avoid unnecessary copies [performance-unnecessary-value-param][0m
                  Mu2eG4SensitiveDetector(name,config) { }
```

Propose to fix these

rlcee, 2023-08-03:
Also just spot-checking some of the low-frequency warnings, they are either potential bugs or at at least good ideas for better code.  So I hope we can start to look at them.

kutschke, 2023-08-31:
I would like to get the warning about a copy that should be a reference into the CI workflow.

rlcee, 2023-08-31:
> I would like to get the warning about a copy that should be a reference into the CI workflow.

It is there, called performance-for-range-copy.  Above, you can see it was found 80 places.  But e28 won't allow it, so it should never happen again after the switch. (or am I forgetting something?)   Also this is the purpose of this Issue, to get clang-tidy into a state where we can reasonably look at the CI tidy result and start enforcing it.  I've been looking recently and it does have good advice.

rlcee, 2023-10-26:
adding -Wextra to the compile flags gives the following warnings counts
     90 [-Werror=deprecated-copy]
     36 [-Werror=extra]
   1377 [-Werror=ignored-qualifiers]
      2 [-Werror=implicit-fallthrough=]
    260 [-Werror=type-limits]
      3 [-Werror=unused-but-set-parameter]
   3287 [-Werror=unused-parameter]

kutschke, 2023-10-26:
Thanks Ray,

   Do you have the log file in a place that I can have a look at it.   I know what most of them are but there are a few I don’t recognize

      Rob

rlcee, 2023-10-26:
> Thanks Ray, Do you have the log file in a place that I can have a look at it.

/exp/mu2e/app/users/rlc/muse/dev/b_03.log

kutschke, 2023-10-26:
> 90 [-Werror=deprecated-copy] 

-  These look like something we should understand and address.  Deprecated could turn into no-longer-suppoerted some day Some will go away when BTrk is commissioned.

> 36 [-Werror=extra] 

-   These are all in BTrk and will go away when BTrk is decomissioned.

> 1377 [-Werror=ignored-qualifiers] 

-  I spot checked these.  The ones that I saw were innocous but I did not check exahaustively

> 2 [-Werror=implicit-fallthrough=] 

- We should just fix these

> 260 [-Werror=type-limits]

- These are actual logic errors in our code and we should ask the authors to address them.

> 3 [-Werror=unused-but-set-parameter] 

- These look like errors in our code and we should ask the authors to address them.

> 3287 [-Werror=unused-parameter]

- 2711  of these occur in 42 .hh files.  We shlould just  fix these.  This leaves us with 560 in 290 .cc files

In summary we should address all except possibly -Werror=ignored-qualifiers and-Werror=unused-parameter that occur in .cc files.  If we decide to address the last two I would split the work up over many people and do it one morning.