Mu2e/Offline #1053: Remove Text Conditions

rlcee, 2023-08-30:
See [this talk](https://mu2e-docdb.fnal.gov/cgi-bin/sso/ShowDocument?docid=46264).

- AcceleratorParams.hh @rlcee (move debauncher to DRPeriod in EventTiming prodition, move proton params to its own fcl)
- CalorimeterCalibrations.hh @sophiemiddleton @bechenard @rlcee (move to recent calo proditions, figure out sim values)
- CalorimeterPhysicalConstants.hh @bechenard @rlcee (remove)
- CrvParams.hh @ehrlich-uva @rlcee (move to header file)
- DAQParams.hh @rlcee (see if it is really used)
- ExtMonFNALConditions.hh  @rlcee @gaponenko  (remove clockticks, new proditions for temp and voltage)

rlcee, 2023-11-17:
- DAQParams is removed
- CrvParams is moved to header
- proton params moved out of AcceleratorParams to fcl

rlcee, 2023-11-17:
- AcceleratorParams removed #1140, last parameter moved to GlobalConstants

sophiemiddleton, 2023-11-21:
I can start to think about the Calo stuff - do we want to meet to discuss that or should I just take a look?

rlcee, 2023-11-21:
> I can start to think about the Calo stuff - do we want to meet to discuss that or should I just take a look?

Yes, I have a little more to do on the others, then we will only be left with the Calo variables.  The summary I showed is in [this doc](https://mu2e-docdb.fnal.gov/cgi-bin/sso/RetrieveFile?docid=46264&filename=compsoft_230705_rmConditions.pdf&version=2), page 5.  CalorimeterPhysicalConstants can probably all be moved to the individual modules.  Then that leaves:

> const std::string& pulseFileName() const {return _pulseFileName;} - simulation, Mu2eUtil
> const std::string& propagFileName() const {return _propagFileName;} - CaloMC
> double LRUpar0(int crystalId) const {return _LRUpar0;} CaloMC one place
> double peMeV(int crystalId) const {return _peMeV;} CaloMC 5 places
> double ADC2MeV(int roId) const {return _ADC2MeV;} Reco - obsolete
> double MeV2ADC(int roId) const {return 1.0/_ADC2MeV;} CaloMC a few places
> double Peak2MeV(int roId) const {return _Peak2MeV;} CaloHitMaker

These can probably be handled by a combination of moving to individual module fcl, GlobalConstants, or the existing Calo proditions.  So I think the next step is to make a plan for each of these.  If it effects Bertrand's code, we should involve him.  We also need to think about not being disruptive, so we don't change product content. If that's necessary, we should probably give a talk at comp/soft.  Can you can deal with any of these easily?  Thanks!

bechenard, 2023-11-21:
Hi everyone,

I worked on some of this yesterday, as I plan to go over the full calo 
code in the coming weeks. Regarding this specific action items:

  - CalorimeterPhysicalConstants is all gone, one problem solved (there 
might be one last item but I have made changes and can make a PR this 
afternoon to fix this).

  - For the remaining conditions (in the list below), we need to discuss 
with the calorimeter people if we want to have something for each 
crystal or if this can be global / semi-global. We also need to add 
support for the LYSO crystal - the tracker has a similar problem and we 
decided we would discuss a common design before rushing forward.

Some of the items listed below will affect several parts of my code, and 
it might be better if I handle this as part of a full calo code 
refactoring.

@Ray, I would like to better understand the interaction with the 
database. Is there some document you could point me to?

Thanks

Bertrand

--------------blI51HAKG6SAE4Yupo89jlN0
Content-Type: text/html; charset=UTF-8
Content-Transfer-Encoding: 8bit

<!DOCTYPE html><html><head>
<meta http-equiv="Content-Type" content="text/html; charset=utf-8">
  </head>
  <body>
    <p>Hi everyone,</p>
    <p>I worked on some of this yesterday, as I plan to go over the full
      calo code in the coming weeks. Regarding this specific action
      items:<br>
    </p>
    <p> - CalorimeterPhysicalConstants is all gone, one problem solved
      (there might be one last item but I have made changes and can make
      a PR this afternoon to fix this).<br>
    </p>
    <p> - For the remaining conditions (in the list below), we need to
      discuss with the calorimeter people if we want to have something
      for each crystal or if this can be global / semi-global. We also
      need to add support for the LYSO crystal - the tracker has a
      similar problem and we decided we would discuss a common design
      before rushing forward.</p>
    <p>Some of the items listed below will affect several parts of my
      code, and it might be better if I handle this as part of a full
      calo code refactoring. <br>
    </p>
    ***@***.***, I would like to better understand the interaction with the
      database. Is there some document you could point me to? </p>
    <p>Thanks</p>
    <p>Bertrand  <br>
    </p>
    <div class="moz-cite-prefix">On 11/21/2023 9:34 AM, Ray Culbertson
      wrote:<br>
    </div>
    <blockquote type="cite" ***@***.***">
      
      <blockquote>
        <p dir="auto">I can start to think about the Calo stuff - do we
          want to meet to discuss that or should I just take a look?</p>
      </blockquote>
      <p dir="auto">Yes, I have a little more to do on the others, then
        we will only be left with the Calo variables. The summary I
        showed is in <a href="https://mu2e-docdb.fnal.gov/cgi-bin/sso/RetrieveFile?docid=46264&amp;filename=compsoft_230705_rmConditions.pdf&amp;version=2" rel="nofollow" moz-do-not-send="true">this doc</a>, page 5.
        CalorimeterPhysicalConstants can probably all be moved to the
        individual modules. Then that leaves:</p>
      <blockquote>
        <p dir="auto">const std::string&amp; pulseFileName() const
          {return _pulseFileName;} - simulation, Mu2eUtil<br>
          const std::string&amp; propagFileName() const {return
          _propagFileName;} - CaloMC<br>
          double LRUpar0(int crystalId) const {return _LRUpar0;} CaloMC
          one place<br>
          double peMeV(int crystalId) const {return _peMeV;} CaloMC 5
          places<br>
          double ADC2MeV(int roId) const {return _ADC2MeV;} Reco -
          obsolete<br>
          double MeV2ADC(int roId) const {return 1.0/_ADC2MeV;} CaloMC a
          few places<br>
          double Peak2MeV(int roId) const {return _Peak2MeV;}
          CaloHitMaker</p>
      </blockquote>
      <p dir="auto">These can probably be handled by a combination of
        moving to individual module fcl, GlobalConstants, or the
        existing Calo proditions. So I think the next step is to make a
        plan for each of these. If it effects Bertrand's code, we should
        involve him. We also need to think about not being disruptive,
        so we don't change product content. If that's necessary, we
        should probably give a talk at comp/soft. Can you can deal with
        any of these easily? Thanks!</p>
      <p style="font-size:small;-webkit-text-size-adjust:none;color:#666;">—<br>
          ID: <span>&lt;Mu2e/Offline/issues/1053/1821364284</span><span>@</span><span>github</span><span>.</span><span>com&gt;</span></span></p>
      <script type="application/ld+json">[
{
"target": "https://github.com/Mu2e/Offline/issues/1053#issuecomment-1821364284",
"url": "https://github.com/Mu2e/Offline/issues/1053#issuecomment-1821364284",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"name": "GitHub",
"url": "https://github.com"
}
}
]</script>
    </blockquote>
  </body>
</html>

--------------blI51HAKG6SAE4Yupo89jlN0--

rlcee, 2023-11-21:
> something for each crystal or if this can be global / semi-global. 

This step does not have to final or anything close.  We only need to move the source of the value.  So, for example, a value could be moved to the exiting Proditions fcl as a constant, and delivered from there, regardless of where it whether it will end up per channel.  But if there is a chance to advance the code without too much delay, then of course, sure.

> I would like to better understand the interaction with the database. Is there some document you could point me to?

at the top of [this wiki page](https://mu2ewiki.fnal.gov/wiki/ConditionsData), there are several links for developers.  But in all cases, I think it is important that I be involved in the plan since I really don't want to waste anyone's time going down the wrong path.  So please once you have a concept outline or any questions, we should meet.

bechenard, 2023-11-21:
Thanks, I'll make an outline of what we want once I have looked at your 
material. I prefer to do it correctly once than in two steps.

--------------Fs7Zyg9s0AfH0t9ngLY9Z2L6
Content-Type: text/html; charset=UTF-8
Content-Transfer-Encoding: 8bit

<!DOCTYPE html><html><head>
<meta http-equiv="Content-Type" content="text/html; charset=utf-8">
  </head>
  <body>
    <p>Thanks, I'll make an outline of what we want once I have looked
      at your material. I prefer to do it correctly once than in two
      steps.</p>
    <p><br>
    </p>
    <div class="moz-cite-prefix">On 11/21/2023 11:20 AM, Ray Culbertson
      wrote:<br>
    </div>
    <blockquote type="cite" ***@***.***">
      
      <blockquote>
        <p dir="auto">something for each crystal or if this can be
          global / semi-global.</p>
      </blockquote>
      <p dir="auto">This step does not have to final or anything close.
        We only need to move the source of the value. So, for example, a
        value could be moved to the exiting Proditions fcl as a
        constant, and delivered from there, regardless of where it
        whether it will end up per channel. But if there is a chance to
        advance the code without too much delay, then of course, sure.</p>
      <blockquote>
        <p dir="auto">I would like to better understand the interaction
          with the database. Is there some document you could point me
          to?</p>
      </blockquote>
      <p dir="auto">at the top of <a href="https://mu2ewiki.fnal.gov/wiki/ConditionsData" rel="nofollow" moz-do-not-send="true">this wiki page</a>,
        there are several links for developers. But in all cases, I
        think it is important that I be involved in the plan since I
        really don't want to waste anyone's time going down the wrong
        path. So please once you have a concept outline or any
        questions, we should meet.</p>
      <p style="font-size:small;-webkit-text-size-adjust:none;color:#666;">—<br>
          ID: <span>&lt;Mu2e/Offline/issues/1053/1821541720</span><span>@</span><span>github</span><span>.</span><span>com&gt;</span></span></p>
      <script type="application/ld+json">[
{
"target": "https://github.com/Mu2e/Offline/issues/1053#issuecomment-1821541720",
"url": "https://github.com/Mu2e/Offline/issues/1053#issuecomment-1821541720",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"name": "GitHub",
"url": "https://github.com"
}
}
]</script>
    </blockquote>
  </body>
</html>

--------------Fs7Zyg9s0AfH0t9ngLY9Z2L6--

bechenard, 2023-12-01:
> CalorimeterCalibrations is now the only remaining text conditions. I'm
> glad to help with any aspect, let me know.

Most, if not all, will need to go into the database. I was planning to 
work on it this week, but I got a few emergencies.

If you want to get it done quickly, we can proceed in two steps by 
moving stuff into another class. Otherwise it might be less work to do 
it properly.

-Bertrand

--------------H1k4P1AHXWb9HH0bUEir0pz6
Content-Type: text/html; charset=UTF-8
Content-Transfer-Encoding: 8bit

<!DOCTYPE html><html><head>
<meta http-equiv="Content-Type" content="text/html; charset=utf-8">
  </head>
  <body>
    <p>Most, if not all, will need to go into the database. I was
      planning to work on it this week, but I got a few emergencies. <br>
    </p>
    <p>If you want to get it done quickly, we can proceed in two steps
      by moving stuff into another class. Otherwise it might be less
      work to do it properly.</p>
    <p>-Bertrand<br>
    </p>
    <div class="moz-cite-prefix">On 12/1/2023 11:48 AM, Ray Culbertson
      wrote:<br>
    </div>
    <blockquote type="cite" ***@***.***">
      
      <p dir="auto">CalorimeterCalibrations is now the only remaining
        text conditions. I'm glad to help with any aspect, let me know.</p>
      <p style="font-size:small;-webkit-text-size-adjust:none;color:#666;">—<br>
          ID: <span>&lt;Mu2e/Offline/issues/1053/1836679937</span><span>@</span><span>github</span><span>.</span><span>com&gt;</span></span></p>
      <script type="application/ld+json">[
{
"target": "https://github.com/Mu2e/Offline/issues/1053#issuecomment-1836679937",
"url": "https://github.com/Mu2e/Offline/issues/1053#issuecomment-1836679937",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"name": "GitHub",
"url": "https://github.com"
}
}
]</script>
    </blockquote>
  </body>
</html>

--------------H1k4P1AHXWb9HH0bUEir0pz6--

rlcee, 2026-02-01:
More progress in PR #1702 #1705 #1706 #1707