Mu2e/Offline #1067: Improve CI parser diagnostics

kutschke, 2023-09-06:
The CI is not smart enough about detecting errors in the command lines it receives.   See, for example, PR #1066 .  There is a command to:

`  run build test with Mu2e/Producton/#265`

This is a mal-formed command that should not have the / before the # .  When the CI parsed this, it simply ignored the entire test with clause and ran the CI without the accompanying change to Production.  So the CI failed.

The job is to improve the CI detect mal-formed commands and give feedback instead of silently doing the incorrect thing.

rlcee, 2023-11-24:
The problem is [here](https://github.com/Mu2e/CI/blob/bcbcaa1a89f46530f89eee6b8ce22920fd7b696d/Mu2eCI/common.py#L93).  The search returns the part that is parse-able and then stops, probably designed to allow a comment to follow the command.  A solution is to require "fullmatch" here which means the full string matches the regex.  This would remove the putative commenting ability unless you wanted to get even trickier.

rlcee, 2023-12-03:
I think this is done, I'll close it tomorrow

rlcee, 2023-12-05:
done with Mu2e/CI#109 and Mu2e/CI#110