Mu2e/Offline #1101: Offspill Crv and Calorimeter noise generation is very inefficient

brownd1978, 2023-10-09:
Generating noise in the calorimeter and crv for 100usec OffSpill events takes ~1 second.  A faster algorithm is needed.

bechenard, 2023-10-11:
I made modifications to speed up the calorimeter noise generation. The noise is generated under "large" energy deposits in each crystal, but the small "salt and pepper" noise is ignored for now (it was already disabled as it requires more work to be done properly, and the physics impact is minimal, especially for off-spill events). My tests indicate that the calo noise takes about 0.01s to generate without any change in the physics w.r.t. the previous code. Should be fine for the time being, and we will work to further improve the noise description (salt and pepper) in the future.

bechenard, 2026-02-12:
I have made changes in the past that have solved this issue for the calorimeter. Leaving the CRV part to Ralf