Mu2e/Offline #1119: charged pion lifetime in ParticleList.txt

pavel1murat, 2023-10-29:
https://github.com/Mu2e/Offline/blob/main/GlobalConstantsService/data/ParticleList.txt#L77  reports the charged pion lifetime of 25.86 ns, different from the PDG average (26.033 ns)

1. that needs to be fixed
2. as it is not clear how one can get such a value, and it is does not look a typo, do all other numbers in ParticleList.txt need to be validated ?

rlcee, 2023-10-29:
To answer your question directly, but not helpfully, when we switched from HepPDT product to the internal ParticleDataList, we intentionally kept all old values so as not to disturb any studies.   The pion width came from [here](https://github.com/Mu2e/Offline/blob/e6ac0514776873c4eddc81fe234ab1e3ee199c71/GlobalConstantsService/data/mass_width_2008.mc#L61), and we simply divide that total width into hbar to get the value in seconds you see in the ParticleDataList file.  This file predates git (and me), so I don't know where it came from, perhaps Rob or Lynn Garren does.  Anyway, I know the history is not the main point.   I'm happy to help with any update or transition the offline group chooses.  The last I heard, the PDG values were not available as csv, so it would be a huge effort to update them all by hand.  Of course, that may have changed, or we can also just update a few low mass particles, or adopt the geant database, which may have its own quirks, but has pion lifetime of 26.033.

pavel1murat, 2023-10-31:
in principle, a 0.5% difference is not a big deal, in this particular case it is just too easy to recognise  ... 
I think the best practical solution would be to check a few numbers we know, make sure that they are correct, and wait till the next 'discovery' 

- perhaps the simulation group also needs to be given a change to voice their opinion

kutschke, 2023-10-31:
I will start a conversation with Andy and Stefano.

kutschke, 2023-10-31:
About the history.  When we started we were told that HepPDT was a standard that had been adopted by the LHC experiments and that it would receive regular updates. So we decided to go with it.  However it was never updated after we started to use it.

brownd1978, 2023-10-31:
PDG is working on a computer-readable form of particle tables, I don't know
their status.  I will check with Richie.  We should migrate to that once
it's available.

sdifalco, 2023-11-27:
Hi everyone, I have just tried to see the differences between the existing Offline/GlobalConstantsService/data/ParticleList.txt and the beta version of machine readable particle data from pdg 
(I used the python interface following  https://pdgapi.lbl.gov/doc/pythonapi.html#particle-monte-carlo-number-mass-and-quantum-numbers) . I have created two python scripts to get the updated particle mass (getParticleMass.py) and lifetime (getParticleTau.py). I have also created a tcsh script ( that uses the two python scripts and the existing  ParticleList.txt  to write an updated ParticleList.txt  that I called NewParticleList.txt. When the PDG information is missing I just copy the old values. The root macro compare_PartList.C compares the old and the new file and writes out the difference in mass or lifetime larger than 0.1%. In principle we can update ParticleList.txt  when we want but the differences are generally small. Once the beta version is more stable we could set up a cleaner automatic procedure to keep the list updated.
[ParticleList.txt](https://github.com/Mu2e/Offline/files/13478793/ParticleList.txt)
[NewParticleList.txt](https://github.com/Mu2e/Offline/files/13478801/NewParticleList.txt)

sdifalco, 2023-11-27:
I had to rename the scripts to attach them here...

sdifalco, 2023-11-27:
[getParticleTau.py.txt](https://github.com/Mu2e/Offline/files/13478832/getParticleTau.py.txt)
[getParticleMass.py.txt](https://github.com/Mu2e/Offline/files/13478833/getParticleMass.py.txt)[updateParticleList.tcsh.txt](https://github.com/Mu2e/Offline/files/13478827/updateParticleList.tcsh.txt)
[compare_PartList.C.txt](https://github.com/Mu2e/Offline/files/13478834/compare_PartList.C.txt)
[ParticleList_diff.txt](https://github.com/Mu2e/Offline/files/13478829/ParticleList_diff.txt)