Mu2e/Offline #1128: Check the work done in PR #1036

kutschke, 2023-11-11:
This issue is about PR #1036 .  Overall it's excellent work and the level of detail in the new STM description is impressive.  But there is one issue for certain and perhaps some others.

The nightly validation that ran early on Nov 8, 2023 caught an unexpected change:

https://mu2e.fnal.gov/atwork/computing/ops/val/valJob/nightly/2023_11/08/pileup/Validation_CrvSteps_compressDetStepMCs_noName_posz.gif

Thanks to Dave for catching this.

This was tracked down to one line PR1036: the volumes:  the two volumes STM_CRVShieldPipe and STM_CRVShieldMatingBlock mating block were turned off in this PR.  These are shielding elements that protect the downstream CRV from albedo off of the STM sweeping magnet and they must be present.

That error will be fixed in an upcoming PR on Sat Nov 11. 

I would like PR #1036 to be checked again by the STM team.  The PR was intended to replace an old version of the STM geometry with a version that is more detailed and a better representation of what will actually be built.  I don't believe that any other changes were intended.

The main thing I would like checked is the list of volume names below.  These are names of volumes that were present before the PR  (SHA bb2a10e5, PR1108)  but are absent after the PR

collimatorSS
collimatorSSliner
ElectronicRackBox_3
mstmDetectorStandLeg1
mstmDetectorStandLeg2
mstmDetectorStandLeg3
mstmDetectorStandLeg4
mstmDetectorStandPlatform
stmDet1
stmDet1Can
stmDet1CanDnStrWindow
stmDet1CanUpStrWindow
stmDet2
stmDet2Can
stmDet2CanDnStrWindow
stmDet2CanUpStrWindow
VirtualDetector_STM_CollDnStr
VirtualDetector_STM_Det1UpStr
VirtualDetector_STM_Det2UpStr
VirtualDetector_STM_SpotSizeCollUpStr

I expect that many or all of the above are part of the old, simple, design that was replaced.  I would like someone from the STM team to sign off on whether or not all of the above volumes were removed intentionally.  

Also, do some need to be added back, perhaps in a revised location, in a later PR?  Here I am thinking of the ElectronicRackBox_3 and the four VirtualDetector_STM volumes.

caohc, 2023-11-12:
Sorry for the inconvenience. I did remove the previous STM volumes on purpose. Because the design of STM Shielding house, Spot-size collimator and STM detectors are all changed.  
     Sorry that I didn't expect that this change would have an impact in the upstream STM area. I will figure out how my changes remove the  STM_CRVShieldMatingBlock and STM_CRVShieldPipe and fix  it that latter.

kutschke, 2023-11-12:
>  I will figure out how my changes remove the STM_CRVShieldMatingBlock and STM_CRVShieldPipe and fix it that latter.

I have already fixed this in PR #1129 .  I understand why you needed to remove volumes that where changed.  I opened this issue for two reasons:

1. To make sure that there were no other accidentally removed volumes.  
2. To make sure that everyone understands which volumes need to be added back in different locations or with different sizes.