Mu2e/Offline #1162: GenId::CeEndpoint vs ProcessCode::CeMinusEndpoint

rlcee, 2023-12-11:
the following is not a bug report, but more of a cleanup proposal.

In the Mu2e MC, there is a certain level of duplication and ambiguity between
GenID's and ProcessCode's, with GenParticle's storing parameter of one type
and the corresponding SimParticle's - parameter of another type, but, in reality,
meaning the same. 

For example, what is the difference between GenId::CeEndpoint  and ProcessCode::CeMinusEndpoint,
(which are not just integer constants but things of different types, so they couldn't be interchanged) ?
of between GenId::externalRPC and ProcessCode::mu2eExternalRPC ?

it seems that it is a consequence of persisting confusion between the notions of a "generator"
and a "generated process" - perhaps that could be sorted  out, as having one type with the meaning
of a generated process code instead of two would make navigating the code base much easier.

-- thanks, regards, Pasha

kutschke, 2023-12-12:
Here is the history.  We can decide that things have evolved and that the type distinction is no longer important but I think that the history is useful input.

In the original conception. GenId was described which generator was run in a generator module.  By construction it was independent of G4.  Each subclass of G4VProcess has a member function that returns a character string that identifies the process.  Originally ProcessCodes were just a mapping of these names to integers for efficiency in storage and comparison time.   We added mu2ePrimary for G4Particles created by PrimaryGeneratorAction, which read the GenParticles from a data product.  

The ProcessCodes and GenIds were made distinct types on purpose so that you could not accidentally compare one to the other.

Then things evolved. First, we added our own code to stop particles mid-G4 for many reasons.  We created ProcessCodes for each of these.  Second, it was inconvenient for people to  discover mu2ePrimary and then have to follow the Ptr to the GenParticle and look up it's GenId.    So someone added some (or all?) of the GenID names to the ProcessCode - the names map to different numbers in the two systems.    I don't know how careful or complete this second step is.   I also don't remember that change being made - I would have objected if I had noticed.

It sounds like there is a request to unifying these two concepts into a single concept.  I am not opposed to this in principle but doing it right will be backwards breaking and intrusive; so it requires planning.

kutschke, 2023-12-12:
To mind my the key question is this: do we have two distinct ideas here or is it really just one idea.  Also, suppose you have a object of the new unified type - will it sometimes be important to know if it's a generator code vs a native G4 code vs a mu2e-addon G4 code?