Mu2e/Offline #1184: g4 testing invokes reconstruction, bringing in database dependency

brownd1978, 2024-02-09:
https://github.com/Mu2e/Offline/blob/c0ec98dc909599388661d13ef65d9c09b7437e73/Mu2eG4/fcl/g4test_03.fcl#L72
shows a dependence of G4 testing on calorimeter reconstruction.  This pulls in the reconstruction condtions database.  Either the reconsruction should be removed, or a default database purpose and version should be provided.

rlcee, 2024-02-09:
@kutschke @resnegfk this is up to you - what do you want tested in this fcl?  (transportOnly too)

kutschke, 2024-02-09:
I am looking at g4test_03MT and transportOnly now.  I suddenly announced divsion meeting at 2:00 PM - hopefully it will be short.

kutschke, 2024-02-09:
Both g4test_03 and transportOnly job developed some mission creep.  I will strip them back to StepPointMCs only.

kutschke, 2024-02-09:
I have a fast and dirty fix for g4test_03 and transportOnly.  I will make a PR now.  It needs addional work - we can discuss to do it in this PR in a separate PR.

rlcee, 2024-02-13:
Addressed in #1185

closed 2024-02-13 by rlcee