Mu2e/Offline #1256: Calorimeter object needs documentation

brownd1978, 2024-05-04:
The Calorimeter geometry object obtained via GeomHandle<Calorimeter> has an opaque interface and lacks documentation.  For instance [CalorimeterGeom/inc/CaloInfo.hh ](https://github.com/Mu2e/Offline/blob/main/CalorimeterGeom/inc/CaloInfo.hh) provides a text-string keyed interface, but there is no documentation what the valid text strings are.  Documentation should be provided in the class headers, and in the appropriate section of the [mu2e wiki](https://mu2ewiki.fnal.gov/wiki/Geometry#Calorimeter), which is currently completely blank.  That should also describe the different geometry systems used in the calorimeter (disk, crystal, etc) and their relationship to each other.

bechenard, 2025-02-14:
I have added stuff in the [wiki](https://mu2ewiki.fnal.gov/wiki/Geometry#Calorimeter) and also inside the code. Do you want anything else?

brownd1978, 2026-02-12:
The wiki now seems to cover this, thanks!
I suggest adding a comment to the effect 'look at the wiki for a list of valid keys" in CaloInfo.hh

bechenard, 2026-02-12:
Just done, PR submitted, closing this

closed 2026-02-12 by bechenard