Mu2e/Offline #1277: Make a mock namespace for  CosmicReco/fcl/prolog.fcl

kutschke, 2024-05-30:
This is to @bonventre @gianipez @rlc and @sophie .  I suggest that we wrap the module definitions inside  CosmicReco/fcl/prolog.fcl a fcl table:

CosmicReco : {
  // existing content
}

and then update fcl references these definitions it to use the new names.  The purpose is to provide some clarity about where definitions came from and to help avoid future name clashes.

@brownd1978 and @oksuzian - if we cover everything in Offline, Production and mu2e_trig_config, is there anything else production related to look at?

sophiemiddleton, 2024-06-10:
I'm not against this idea. I think @bonventre can say more about how much other code now depends on this .fcl defintion. As far as I recall, CosmicReco is fairly self contained so might be a good starting point for this

bonventre, 2024-06-10:
Sounds good to me. I will also change TrackerAlignment fcl to match

edcallaghan, 2024-06-27:
FYI, I believe this was implemented by @gianipez in https://github.com/Mu2e/Offline/pull/1286. So @bonventre you may want to update `TrackerAlignment` now.

@sophiemiddleton When I started down this road it did propagate to the `Production` fcls. There were uses of definitions in a previously-defined `CosmicTracking` namespace, which it looks like Giani changed to `CosmicReco` --- so that will need to be updated.