Mu2e/Offline #1352: Remove documenting histograms from Generators to an accompanying analyzer module

kutschke, 2024-10-04:
I the early days of Mu2e we wrote event generator modules to include histotgrams that document the generator's actions.  We now prefer separate modules - a clean generator module and an accompanying analyzer module.  The reason is that this makes it possible to run just the analyzer module and point it at an existing data product.

There are some event generators that still are organized the old way.  One example is:

https://github.com/Mu2e/Offline/blob/main/EventGenerator/src/RPCGun_module.cc

As I discover more I will edit this comment to add them (and to mark then as resolved when they are resolved ).

sophiemiddleton, 2024-10-04:
Hi Rob, I believe we have generic plotting module: https://github.com/Mu2e/Offline/blob/main/EventGenerator/src/GeneratorPlots_module.cc . Perhaps this can be re-purposed for the RPC too.

kutschke, 2024-10-04:
I think that will work.  It's an extra step to make plots from the TTree but that's easy to do and more flexible that making plots in code.  Please remove the histograms from RPCGun_modules.  I will merge #1341  now and you can start a new PR to make this change.

sophiemiddleton, 2024-10-04:
Thanks, I will put in another PR soon with the plots removed